guysoft/OctoPi #596: USB camera muddled up on re-boot.

usk-johnny-s, 2019-06-25:
#### What were you doing?

I'm using multiple USB WebCam.

#### What did you expect to happen?

I wish to fixed assignment of camera of each stream.

#### What happened instead?

But, RaspberryPi 's  "/dev/video#" assignment changes some-time on a re-boot.
I would like to specify a camera device with "-d /dev/v4l/*" in "/boot/octpi.txt"

Ref: https://lb.raspberrypi.org/forums/viewtopic.php?t=70274

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPi

0.16.0

#### Printer model & used firmware incl. version

Ender-3 with Marlin 1.1.9

#### Screenshot(s)/video(s) showing the problem:

Nothing

usk-johnny-s, 2019-06-25:
Could you check my diff?

https://github.com/usk-johnny-s/OctoPi/commit/7cec9256e2f228b5abccbc913a9452c75f9e3d43#diff-22aa2395ed564e7479b3bdb97786dbd8

guysoft, 2019-07-02:
Fixed merged

closed 2019-07-02 by guysoft