guysoft/OctoPi #598: Microsoft VX-7000 webcam: additional_brokenfps_usb_devices

numanair, 2019-06-27:
#### What were you doing?

Using a Microsoft VX-7000 webcam, removing the -f parameter was required to get an image.

lsusb reports the ID: 045e:0723

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPi

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

guysoft, 2019-06-27:
Can you please provide more details? You left all the rest and have no clue what went wrong, what version are you using and so on.

numanair, 2019-06-28:
My apologies, I assumed the other sections were not needed in this case.
I have tested against Octopi 0.14 through 0.16.
The webcam works fine, it seems it just needs to be added to additional_brokenfps_usb_devices or have the -f parameter removed.

guysoft, 2019-06-28:
There is a page documenting that you can update here:
https://github.com/foosel/OctoPrint/wiki/Webcams-known-to-work

guysoft, 2019-06-28:
@foosel Is the the place to update? Right?
https://github.com/foosel/OctoPrint/wiki/Webcams-known-to-work

foosel, 2019-07-01:
Currently yes.

guysoft, 2019-07-02:
Ok, closing this @numanair upldate it in the wiki page

closed 2019-07-02 by guysoft