guysoft/OctoPi #599: Armbian image not found

Lino77, 2019-07-07:
take a look at the build.log the armbian image is in /OctoPi/src/image-variants and they could not find image.
the 
[build.log](https://github.com/guysoft/OctoPi/files/3365915/build.log)

guysoft, 2019-07-08:
What is the name of the base image you are building with?
You need to download an image and place it in the ``image-variants`` folder

Lino77, 2019-07-08:
the name of teh image is 
Armbian_5.90_Tinkerboard_Debian_buster_default_4.4.182.7z
and it is in Octopi/src/image-variants

guysoft, 2019-07-09:
There was a bug, @Lino77 try pulling latest dev CustomPiOS and try again. tell me if it fails.

closed 2019-07-09 by guysoft

Lino77, 2019-07-09:
okay image where found but it dosn t build see the build.log 
[build.log](https://github.com/guysoft/OctoPi/files/3374308/build.log)

guysoft, 2019-07-10:
ok, pushed another fix, try now

Lino77, 2019-07-10:
no sorry 
[build.log](https://github.com/guysoft/OctoPi/files/3378633/build.log)

guysoft, 2019-07-10:
Ok, seems like permission issues with the apt-cache, I disabled it now, try using latest CustomPiOS from devel branch and built with it

Lino77, 2019-07-11:
Sorry,
[build.log](https://github.com/guysoft/OctoPi/files/3382874/build.log)
Image is still not built

guysoft, 2019-07-11:
Commit reference: https://github.com/guysoft/CustomPiOS/commit/dd2c12edabaa9a58859c702f6a68f3f825812303
@Lino77 Thats an issue with Debian/Armbian:
https://forum.armbian.com/topic/10875-info-checkstart-manually-apt-update-because-of-buster-repository-value-changes/

Its there because they didn't upgrade the base image to buster correctly.

I pushed a fix for that, try now, thanks for keeping up

Lino77, 2019-07-11:
okay, because I have again, the next mistake, I think I'm still waiting a few weeks to build the image. think that the error now all comes from the Arbian image

sed -i 's@ExecStart=/usr/bin/ssh-keygen -A -v@ExecStart=/bin/bash -c '\'' /usr/bin/ssh-keygen -A -v >> /var/log/regenerate_ssh_host_keys.log 2>\&1'\''@g' /lib/systemd/system/regenerate_ssh_host_keys.service
/bin/sed: can't read /lib/systemd/system/regenerate_ssh_host_keys.service: No such file or directory
++++ echo_red 'build failed, unmounting image...'
++++ echo -e -n '\e[91m'
++++ echo build failed, unmounting image...
build failed, unmounting image...
++++ echo -e -n '\e[0m'

[build.log](https://github.com/guysoft/OctoPi/files/3383312/build.log)

guysoft, 2019-07-17:
@Lino77 Yes, its building, its failing because the Armbian image does not have regenerate_ssh_host_keys.service.  Committed a fix, pull and build again.
Each commit here is a step to get this fully working.

Lino77, 2019-07-18:
Okay the next one 

 Downloading https://files.pythonhosted.org/packages/5c/e0/be401c003291b56efc55aeba6a80ab790d3d4cece2778288d65323009420/pip-19.1.1-py2.py3-none-any.whl (1.4MB)
Installing collected packages: pip
  Found existing installation: pip 9.0.1
    Not uninstalling pip at /usr/lib/python2.7/dist-packages, outside environment /usr
Successfully installed pip-19.1.1
+ pip install virtualenv --upgrade
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    from pip import main
ImportError: cannot import name main
+ exit 1
Connection to 127.0.0.1 closed.
lino@ubuntu:~/tinker/OctoPi/src/vagrant$ 

[build.log](https://github.com/guysoft/OctoPi/files/3408403/build.log)

guysoft, 2019-08-05:
@Lino77 Took a while because I was on vacation. 
Fixed, try now. the issue is that pip was installed twice

Lino77, 2019-08-10:
2019-08-10 08:48:03 (5.28 MB/s) - written to stdout [1775087/1775087]

Traceback (most recent call last):
  File "<stdin>", line 22308, in <module>
  File "<stdin>", line 197, in main
  File "<stdin>", line 82, in bootstrap
  File "/tmp/tmpvx3y102y/pip.zip/pip/_internal/__init__.py", line 40, in <module>
  File "/tmp/tmpvx3y102y/pip.zip/pip/_internal/cli/autocompletion.py", line 8, in <module>
  File "/tmp/tmpvx3y102y/pip.zip/pip/_internal/cli/main_parser.py", line 7, in <module>
  File "/tmp/tmpvx3y102y/pip.zip/pip/_internal/cli/cmdoptions.py", line 15, in <module>
ModuleNotFoundError: No module named 'distutils.util'
+ exit 1
Connection to 127.0.0.1 closed.
lino@ubuntu:~/tinker/OctoPi/src/vagrant$
[build.log](https://github.com/guysoft/OctoPi/files/3488689/build.log)

guysoft, 2019-08-11:
@Lino77 Ok, fixed that.

Lino77, 2019-08-11:
sudo: unable to change expired password: Authentication token manipulation error
++++ echo_red 'build failed, unmounting image...'
++++ echo -e -n '\e[91m'
++++ echo build failed, unmounting image...
build failed, unmounting image...
++++ echo -e -n '\e[0m'
++++ cd /distro
++++ unmount_image /distro/workspace-armbian/mount force
++++ mount_path=/distro/workspace-armbian/mount
[build.log](https://github.com/guysoft/OctoPi/files/3489936/build.log)