guysoft/OctoPi #600: avconv is not available in buster

JirkaChadima, 2019-08-16:
#### What were you doing?

1. Run a print with a timelapse set
2. Print ends, timelapse creation crashes with 255 Unknown Error

#### What did you expect to happen?

Get a timelapse

#### What happened instead?

No timelapse. Getting the following error in octoprint.log

```
2019-08-16 09:54:41,183 - octoprint.timelapse - ERROR - Could not render movie due to unknown error
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/timelapse.py", line 833, in _render
    p = sarge.run(command_str, stdout=sarge.Capture(), stderr=sarge.Capture())
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1462, in run
    p.run(input=input, async_=async_)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1071, in run
    self.run_node(node, input=input, async_=False)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1187, in run_node
    result = getattr(self, method)(node, input, async_)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1333, in run_command_node
    node.cmd.run(input=input, async_=async_)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 658, in run
    raise ValueError('Command not found: %s' % self.args[0])
ValueError: Command not found: /usr/bin/avconv
```

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPi

Fully updated/upgraded nightly build for RPi4, 0.17.0

#### Printer model & used firmware incl. version

Prusa MK3S

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

JirkaChadima, 2019-08-16:
It's probably due to libav-tools getting dropped from Buster, I'll check if timelapse gets created with `ffmpeg` instead of `avconv`.

guysoft, 2019-08-18:
Ok, can you write in detail what packages you installed and what configuration you changed for it to work.

Also cc @foosel this means timelapse is broken in nightly

JirkaChadima, 2019-08-19:
Sure. I tried to install `ffmpeg` package, but it turns out it's already installed in the factory octopi image. So all I did in the end was to change the configuration in OctoPrint's UI from `/usr/bin/avconv` to `/usr/bin/ffmpeg`

Octopi is setting the path [here](https://github.com/guysoft/OctoPi/blob/master/src/modules/octopi/filesystem/home/pi/.octoprint/config.yaml#L4) and I was not able to find any mention of `avconv` in octoprint's `master` branch. I think the fix for buster would be to change the path in octopi's build config.

guysoft, 2019-10-23:
This should be fixed in nightly, anyone can confirm?

JirkaChadima, 2019-10-23:
The one from October 13 on [this page](https://gnethomelinux.com/OctoPi/nightly/)? I can confirm, but in a few days time, I'm a little busy ATM.

guysoft, 2019-10-23:
@JirkaChadima Yes that should include the fix, confirm when you can :)

JirkaChadima, 2019-10-24:
With 

2019-10-13_2019-07-10-octopi-buster-lite-0.17.0.zip | 2019-10-13 00:25:06 UTC | 661.3MB | 95b094797cd526455cbb6f57fefb7940
-- | -- | -- | --

I'm still seeing avconv preconfigured. So I guess the next nightly after today's PR merge will contain the fix?

guysoft, 2019-10-24:
It was merged 2 hours ago. Next nightly.

closed 2019-10-24 by guysoft

guysoft, 2019-10-24:
Yes