guysoft/OctoPi #602: additional_brokenfps_usb_devices=("038f:6001")

hubione, 2019-08-19:
#### What were you doing?
As writed on /boot/octopi.txt: "If this fixes your problem, please report it back so we can include the device out of the box: https://github.com/guysoft/OctoPi/issues"
So I write to you, to add my webcam on your source code.

1. connect usb beroscope china cam
2. add line additional_brokenfps_usb_devices=("038f:6001") to /boot/config.txt
3. it works

#### What did you expect to happen?
work without modifying octopi.txt
#### What happened instead?
no video
#### Did the same happen when running OctoPrint in safe mode?
Not tested

#### Version of OctoPi
Version 1.3.11

#### Printer model & used firmware incl. version
prusa i3 mk3s, latest firmware

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

guysoft, 2019-08-20:
Hey, TO make sure I understand - you are using "Chinese Endoscope 8mm" camera and its not working?

hubione, 2019-08-20:
Yes, I put the beroscope near the head to have a close up of the print.
**Without** line (additional_brokenfps_usb_devices=("038f:6001") on /boot/config.txt) there is **NO** video.
Sorry for the doubt

guysoft, 2019-08-20:
The reporter that it works is @michalzielanski which might have any more input.

closed 2019-08-20 by guysoft

michalzielanski, 2019-08-21:
My version of the camera works with and without the line `additional_brokenfps_usb_devices=("038f:6001")` in `/boot/octopi.txt`.

Detailed information about my camera:
```
pi@octopi:~ $ sudo service webcamd stop
pi@octopi:~ $ sudo lsusb -D /dev/bus/usb/001/005 
Device: ID 038f:6001  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x038f 
  idProduct          0x6001 
  bcdDevice            5.38
  iManufacturer           1 lihappe8 Corp.
  iProduct                2 USB 2.0 Camera
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          629
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               4 USB 2.0 Camera
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0 
      iInterface              4 USB 2.0 Camera
      VideoControl Interface Descriptor:
        bLength                13
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.00
        wTotalLength           79
        dwClockFrequency       30.000000MHz
        bInCollection           1
        baInterfaceNr( 0)       1
      VideoControl Interface Descriptor:
        bLength                28
        bDescriptorType        36
        bDescriptorSubtype      6 (EXTENSION_UNIT)
        bUnitID                 6
        guidExtensionCode         {b0d0bb68-a461-834b-90b7-a6215f3c4f70}
        bNumControl            24
        bNrPins                 1
        baSourceID( 0)          2
        bControlSize            3
[... 391 lines cut ...]
      bInterfaceSubClass      2 Video Streaming
      bInterfaceProtocol      0 
      iInterface              4 USB 2.0 Camera
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            5
          Transfer Type            Isochronous
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)
```

hubione, 2019-08-22:
Hello,
I test again.

additional_brokenfps_usb_devices=("038f:6001") IN /boot/octopi.txt WORKS
remove line and restart octoprint (not reboot) WORKS
reboot (without additional_brokenfps_usb_devices=("038f:6001") IN /boot/octopi.txt) DIDN'T WORK

So I can confirm for me, line additional_brokenfps_usb_devices=("038f:6001") IN /boot/octopi.txt is needed 

Hub

guysoft, 2019-08-25:
been merged and available in the nightly

commit 6d08ff02 (2019-08-20): Fixes #602 Chinese Endoscope 8mm