guysoft/OctoPi #603: Orange Pi Zero H5 (qemu: Unsupported syscall: 277)

uNbAs, 2019-09-01:
Good morning

I am trying to do the port of octprint for the OrangePi Zero and I encounter some problems.

The steps I have followed have been a bit diverse. I started cloning the CustomPIOS repo but I realized that there are quite a few branches, so as I wanted to make an armbian image I have based on:
https://codeload.github.com/guysoft/OctoPi/zip/variantsandflavors

I have changed the use of qemu-arm-static for qemu-aarch64-static but it gives me an error in a syscall call:

Setting up libavahi-common-data: arm64 (0.7-4 + b1) ...
Setting up libavahi-common3: arm64 (0.7-4 + b1) ...
Setting up libavahi-core7: arm64 (0.7-4 + b1) ...
Setting up avahi-daemon (0.7-4 + b1) ...
You are required to change your password immediately (administrator enforced)
chfn: PAM: Authentication token is no longer valid; new one required
adduser: `/ usr / bin / chfn -f Avahi mDNS daemon avahi 'returned error code 1. Exiting.
dpkg: error processing package avahi-daemon (--configure):
 installed avahi-daemon package post-installation script subprocess returned error exit status 1
Processing triggers for systemd (241-5) ...
Processing triggers for man-db (2.8.5-2) ...
qemu: Unsupported syscall: 277

[build.log](https://github.com/guysoft/OctoPi/files/3563352/build.log)

guysoft, 2019-09-01:
1. The code you linked is from 2015, please use the current codebase in this Repo.
2. What image are you using? It can't determine the architecture of the image without knowing which image it is.

uNbAs, 2019-09-01:
First I tried to do it with the branch of CustomPIOS but it did not reach any point, although I did ./build armbian did not get to mount the image or work on it.

I have renamed the downloaded image to contain Armbian's name - and match the search pattern, but the only branch that seems to work is that of variantsflavors. Maybe I have done something wrong in the CustomPIOS although I have followed the steps indicated in the readme

This links are download imagen for armbian
https://www.armbian.com/orange-pi-zero-2-h5/
https://dl.armbian.com/orangepizeroplus2-h5/Debian_buster_next.7z

guysoft, 2019-09-01:
CustomPiOS is not a branch, its a library used. Please provide where the CustomPiOS build failed. The code you posted above is legacy.

uNbAs, 2019-09-01:
I have tried, with the instructions of the Readme of CustomPIOS, I have modified in the host system the same as the guy in issue # 23 did:
mv / usr / bin / qemu-arm-static / usr / bin / qemu-arm-static_save
ln -s / usr / bin / qemu-aarch64-static / usr / bin / qemu-arm-static

Too I changed line for search * raspbian - *. Zip for * .zip since I only have one compressed image in the directory

But fail in:
+ cp -v -r --preserve = mode, timestamps / filesystem / root /. /
/common.sh: line 132: / bin / cp: No such file or directory
+ exit 1
[build.log](https://github.com/guysoft/OctoPi/files/3563389/build.log)

uNbAs, 2019-09-01:
In this case I used the OpiZero H5 itself but it still fails, although it advances more than the other host

> 
> + sudo apt-get -y install policykit-1
> sudo: Account or password is expired, reset your password and try again
> Changing password for root.
> sudo: no tty present and no askpass program specified
> sudo: unable to change expired password: Authentication token manipulation error
> ++++ echo_red 'build failed, unmounting image...'
> ++++ echo -e -n '\e[91m'
> ++++ echo build failed, unmounting image...
> build failed, unmounting image...
> ++++ echo -e -n '\e[0m'
> ++++ cd /root/OctoPi/src
> ++++ unmount_image /root/OctoPi/src/workspace-armbian/mount force
> ++++ mount_path=/root/OctoPi/src/workspace-armbian/mount
> ++++ force=
> ++++ '[' 2 -gt 1 ']'
> ++++ force=force
> ++++ '[' -n force ']'
> +++++ sudo lsof /root/OctoPi/src/workspace-armbian/mount
> +++++ awk '{print $2}'
> +++++ sudo mount
> +++++ grep /root/OctoPi/src/workspace-armbian/mount
> +++++ awk -F ' on ' '{print $2}'
> +++++ awk '{print $1}'
> +++++ sort -r
> ++++ for m in $(sudo mount | grep $mount_path | awk -F " on " '{print $2}' | awk '{print $1}' | sort -r)
> ++++ echo 'Unmounting /root/OctoPi/src/workspace-armbian/mount/dev/pts...'
> Unmounting /root/OctoPi/src/workspace-armbian/mount/dev/pts...
> ++++ sudo umount /root/OctoPi/src/workspace-armbian/mount/dev/pts
> ++++ for m in $(sudo mount | grep $mount_path | awk -F " on " '{print $2}' | awk '{print $1}' | sort -r)
> ++++ echo 'Unmounting /root/OctoPi/src/workspace-armbian/mount/dev...'
> Unmounting /root/OctoPi/src/workspace-armbian/mount/dev...
> ++++ sudo umount /root/OctoPi/src/workspace-armbian/mount/dev
> ++++ for m in $(sudo mount | grep $mount_path | awk -F " on " '{print $2}' | awk '{print $1}' | sort -r)
> ++++ echo 'Unmounting /root/OctoPi/src/workspace-armbian/mount...'
> Unmounting /root/OctoPi/src/workspace-armbian/mount...
> ++++ sudo umount /root/OctoPi/src/workspace-armbian/mount
> ++++ echo_red -e '\nBUILD FAILED!\n'
> ++++ echo -e -n '\e[91m'
> ++++ echo -e '\nBUILD' 'FAILED!\n'
> 
> BUILD FAILED!
> 
> ++++ echo -e -n '\e[0m'
> + exit 1

attached build.log file

guysoft, 2019-09-01:
It says that your image does not have the command ``cp`` which is a basic POSIX command. 
aka this is really strange behaviour. Because there is cp in that image.

I think I can see your issue.
```
 ln -s / usr / bin / qemu-aarch64-static / usr / bin / qemu-arm-static
```

You can't symlink this because as the command run it goes in to a chroot, which would make the symlink invalid. A chroot changes the root and therefore symlinks will not work because the absolute path would change.
You need to have a copy of the file there.

Side note: You are renaming ``qemu-aarch64-static`` to ``qemu-arm-static``. That could have been a potential for havoc. But surprisingly both work.

uNbAs, 2019-09-02:
As you say, the problem is that when the chroot comes into play, the "trick" of the symbolic link is lost.

But I already have the logic created to solve it. As soon as I do some tests and compile the image correctly, I send you the data in a request.

uNbAs, 2019-09-03:
Hi! Please close thi issue

closed 2019-09-03 by uNbAs