guysoft/OctoPi #604: [Request] OctoPi image with Python3 out-of-the-box

OutsourcedGuru, 2019-09-04:
#### What were you doing?
Like many of us, I'm trying to work with Python 3 to make sure that my plugins will weather the test of time.

So I setup a standard OctoPi-imaged rig, then basically:

1. `mv ~/oprint ~/oprint.py2`
2. `virtualenv -p python3 oprint`
3. `source oprint/bin/activate`
4. `pip install --upgrade pip`
5. `pip install https://get.octoprint.org/latest`
`ERROR: Package 'OctoPrint' requires a different Python: 3.7.3 not in '>=2.7.9,<3'`

Interesting fact: For whatever reasons, I've gone through this routine perhaps five times already, testing different scenarios and saving each to their own microSD.

#### Request
What might be nice is a turnkey OctoPi image which pulls the latest Buster Lite as well as the latest OctoPrint 1.4rc_something_ from the `devel` branch. In theory, it would still have the classic `~/oprint` as the virtual environment location, just with Python 3.7.3 instead of Py2 throughout. Kudos if it's Pi4-friendly.

So basically—and with some instructions for how to pull this—one would:

1. Use Etcher to pull from some URL this IMG
2. Do the standard `/boot/octopi-wpa-supplicant.txt` thing
3. Boot it up

Remoting in would reveal a vanilla OctoPrint installation, just now Python 3—savvy.

I would suggest that this would benefit people in the plugin development space in the interim period before OP 1.4 is released.

#### What did you expect to happen?
I'd forgotten that we don't get Python 3 support until 1.4 which isn't released to `master` yet. But I suppose I naively expected this to work.

#### What happened instead?
The scenario requires the would-be tester to be able to install some as-yet-released OctoPrint version on the `devel` branch.

#### Did the same happen when running OctoPrint in safe mode?
N/A

#### Version of OctoPi
Presumably it would have been 1.3.11 at that point for OctoPrint and this would have been the daily build which includes Buster. 

#### Printer model & used firmware incl. version
N/A

#### Screenshot(s)/video(s) showing the problem:
N/A

I have read the FAQ.

jonasbits, 2019-09-05:
I would also like this very much! Hopefully i can help to convert some plugins to support py3

guysoft, 2019-09-05:
I could make a a python3 branch - @foosel is it time to start setting this stuff up?

OutsourcedGuru, 2019-09-08:
I don't want to rush anyone but neither Kivy nor QT5 work now on the Pi4B. A work-around in the Kivy space can be found using a Python3 path but they've now discontinued any work whatsoever for Python2; so there will be no working Python2+Kivy+4B solution available.

guysoft, 2019-09-08:
Why do you need kivy?

>

OutsourcedGuru, 2019-09-08:
The OctoPrint plugin I'm developing is for a really amazing 3D printer which includes a 10" TFT display. The display interface is written in Kivy.

For what it's worth, here's a link to a [post I just made on one of the Kivy issues](https://github.com/kivy/kivy/issues/6413#issuecomment-529256007).

guysoft, 2019-09-09:
Thats neat!
Its always good to link between issues.
Been playing with kivy for android, but never got something I was happy with using.

OutsourcedGuru, 2019-09-09:
There's a great support community of theirs on Discord if you ever wanted to see the Android build process.

lpla, 2020-07-02:
Hi. Any progress on this? Python 2 support ended six months ago.

PR #658 (2020-07-03): Switch to Python 3
Switches OctoPrint's venv to Python 3

Closes #604
commit 7f6170ad (2020-07-03): Switch to Python 3

Closes #604
closed 2020-07-03 by guysoft