guysoft/OctoPi #606: System clock out of sync on boot

seltix5, 2019-09-09:
#### What were you doing?

Original issue created here https://github.com/foosel/OctoPrint/issues/3266, it was closed because it's a Octopi issue, not Octoprint.

Already tested this on a new instalation with no plugins installed.
And NTP is working :
```
pi@octopi-ender3:~ $ timedatectl status
      Local time: Sun 2019-09-08 19:39:31 WEST
  Universal time: Sun 2019-09-08 18:39:31 UTC
        RTC time: n/a
       Time zone: Europe/Lisbon (WEST, +0100)
 Network time on: yes
NTP synchronized: yes
 RTC in local TZ: no
```

1. Cut power off from raspberry pi ( without using the Shutdown system from UI )
2. wait something like 5 or 10min
3. Boot up raspberry pi
4. Check octoprint.log, the timestamp will be equal or INFERIOR at timestamp before shutdown

[octoprint.log](https://github.com/guysoft/OctoPi/files/3589529/octoprint.log)
[serial.log](https://github.com/guysoft/OctoPi/files/3589532/serial.log)

#### What did you expect to happen?

I wold expect octoprint to wait for NTP sync but I already understod it may not be possible. I already understood too because raspberry pi has not hardware clock with battery is not possible to maintain the clock updated at start up, only after online sync with NTP server. But I wold expect to at least the time stamp wont be inferior to the one at shutdown.
For example at `2019-09-08 19:08:50,449` octoprint startup with a timestamp smaller than before shutdown.

#### What happened instead?

Some times the boot up timestamp is inferior to shutdown timestamp.

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPi

OctoPrint 1.3.11 on OctoPi 0.16.0

#### Printer model & used firmware incl. version

Ender 3 with original Marlin 1.1.6

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

guysoft, 2019-09-10:
I am not really sure how to debug this. OctoPi is basically built on top of Raspbian. I am not really feeling comfortable sending you to yet another issue tracker. But I am not really changing anything in the timeszone settings from there. 
As a workaround you could just get an RTC.

There are $1 ones on aliexpress, and a really neat one which would need no wiring by adaftuit for $10.
https://pimylifeup.com/raspberry-pi-rtc/

seltix5, 2019-09-10:
oh.. I was hopping there was something to debug :S I can report the issue in another place, where should I do that?
This really is a strange situation with the time clock going back, it probably is software related and it should be possible to fix it.

guysoft, 2019-09-10:
https://www.raspbian.org/RaspbianBugs

They seem to have a launchpad:
https://bugs.launchpad.net/raspbian

seltix5, 2019-09-10:
I will report there, thank.
I will update if I get any evolution in this problem.
thanks

seltix5, 2019-09-10:
reported : https://bugs.launchpad.net/raspbian/+bug/1843398