guysoft/OctoPi #607: ..

3Dangel, 2019-09-13:
#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPi

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

guysoft, 2019-09-15:
This issue is empty.
Closing if no reply.

closed 2019-09-15 by guysoft

3Dangel, 2019-09-15:
ooops, sorry for troubling here...

I was trying to get my webcam and Elo Touch screen to work and trying to search as many docs as I could...
not sure what brought me into this issue report...

I will report properly if issue with my webcam.... hahah still don't want to install 

something was missing from the repo...."  ffmpeg is not available"

pi@raspberrypi:~ $ sudo apt install subversion libjpeg62-turbo-dev imagemagick f                                                                                                                                                             fmpeg libv4l-dev cmake
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package ffmpeg is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

guysoft, 2019-09-16:
The template there is there for a reason.
I don't know what version of OctoPi, and there were changes between version because of Debian.
You are installing ``-dev`` pacakges which would make no sense unless you are compiling something, so you are not explaining what guide you are following and why you are doing that.
Also you didn't say which camera model. 
So, this information alone is still useless I am afraid. 
Once you have all the details then you can report it neatly

3Dangel, 2019-09-18:
What were you doing?
1-installing webcam package
2- instruction source from
 https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337
3- instruction code:
sudo apt install subversion libjpeg62-turbo-dev imagemagick ffmpeg libv4l-dev cmake

What did you expect to happen?
I was expecting the package to be installed correctly 

What happened instead?
caused error reporting missing or source been deleted:
**...Package ffmpeg is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source"**

Did the same happen when running OctoPrint in safe mode?
it does not affect operation of the OctoPrint
Version of OctoPrint
1.3.11

Printer model & used firmware incl. version
Zonestar 802 Prusa i3 clone
Mks-Robin V1.03 (close source)
Screenshot(s)/video(s) showing the problem:

[image]

thanks and sorry for been newbie here...

guysoft, 2019-09-19:
What version of OctoPi, not OctoPrint only, you still didn't answer this

>

3Dangel, 2019-09-19:
I did not use octopi image. 
There is no octopi.txt. 
My issue , I think is the changes at targeted source either the file name or something like that. 
Really admire your work!!!
And THANK you for your support!!!

I have spent another 16 hs( well, newbieis my name) to find and try piece by piece. 

ffmpeg is installed directly from it source on GitHub. 
Then complied the rest of instructions

Now manually, I can lunch the mpeg_stream and using my pi’s ip:8080 I can see and take picture. 

I was almost about to give up. 

My issue is not using the provided octopi’s SD card image. That should have been plug&play...I think. 

But, since I have an Elo 10” touch screen to use, I needed it to be mounted first under raspbian Pixel and then install octoprint. 

But, I learned quite a lot!
Thanks again for your help!!
Appreciated the time you have put into the whole project!!!!! Please keep up the great job!!!!!!!!!

guysoft, 2019-09-19:
I am not sure what you are trying to say. 
You are not using OctoPi but you want to tell me its good? Then why aren't you using it?

3Dangel, 2019-09-19:
:-)
May be it is the terminology that I’m not familiar with.
PI model 3b
1st stage
Installed raspbian pixel
2nd stage
Install touch screen
3 stage
Install Octoprint
Followed instruction from
https://octoprint.org/download/
Installing from source

4stage
Installing webcam

So, by installing the OctoPrint, the pi become OctoPi?

Right now, my Pi is on all the time, I ssh from Win10 PC and do the
installation this way from stage 2 onward. Sometime I needed to directly
use the pi to get install/tests done. ( mainly for the Elo touchscreen)

I have successfully connected Octoprint with my 3d printer and printed that
way.
Accessing and uploading gcidenfile from my Win 10 pc
Dialed into Octoprint via pi’s IP:5000 port

So , as user, it is working.
:-)

guysoft, 2019-09-22:
> So, by installing the OctoPrint, the pi become OctoPi?

No, OctoPi is a distrubition for the RaspeberryPi, if you want to use OctoPi you have to use the distribution, or use the build scripts to generate an image that uses it.
See: https://en.wikipedia.org/wiki/Linux_distribution

3Dangel, 2019-09-22:
Thanks a lot!!!
I will be building another pi, so it will from the image from OctoPi.
Thanks again for the help.