guysoft/OctoPi #608: Package ffmpeg is not available..... ?

3Dangel, 2019-09-15:
my webcam.... hahah still don't want to install

something was missing from the repo...." ffmpeg is not available"

background:
install Rpi raspbian with GUI and Chrom
manual installation of Octoprint via Github instruction
all works but webcam...
any idea where I should be looking into?
I'm at kindergarten level in respect of linux environment.... :-) learning to walk still

thanks in advance!!
....

pi@raspberrypi:~ $ sudo apt install subversion libjpeg62-turbo-dev imagemagick f fmpeg libv4l-dev cmake
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package ffmpeg is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

guysoft, 2019-09-16:
Please fill in the template - what version of OctoPi. See other comment why I am asking this: https://github.com/guysoft/OctoPi/issues/607#issuecomment-531720048

3Dangel, 2019-09-16:
> why I am asking this: #607 (comment)

Will do.
Thanks a lot for your feedback.

closed 2019-09-20 by 3Dangel