guysoft/OctoPi #610: REQUEST: Support for ZiggyHD webcam

MrTeacherMan, 2019-10-23:
#### What were you doing?

Connected Ziggi HD webcam (UVC) it works but when i attempt to control auto-focus via ucvdynctrl 

#### What did you expect to happen?
Should have been able to adjust the focus of the webcam
#### What happened instead?
it ends up in a loop because v4l2 is "buggy" (actual term used in the error)
#### Did the same happen when running OctoPrint in safe mode?
was in SSH to terminal

#### Version of OctoPi

OctoPi 0.16.0

#### Printer model & used firmware incl. version
 N/A

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

POSSIBLE SOLUTION:

https://mikeh69.wordpress.com/ipevoziggihd/

I have attempted the solution, but get the error:

"Patch input contains nothing but Garbage"

guysoft, 2019-10-24:
Does it work in any form on Rasbian?

MrTeacherMan, 2019-10-24:
here is a video that the manufacturer posted about using ziggyHD on a Pi:

https://twitter.com/ipevo/status/830136773355331585

guysoft, 2019-10-24:
Thats from 2017. And has no technical detail. It might not be supported
anymore.

If it doesn't work out of the box on vanilla raspbian and there is no
driver information I have no way to debug this.

MrTeacherMan, 2019-10-24:
My personal Linux experience is very little, from what I know Raspbian is a derivative of Debian Linux.

Would not the first fix I linked, from Mike H that corrects the error in libwebcam being unable to jump to the next UVC driver work, if the patch file would load properly?

guysoft, 2019-10-28:
@MrTeacherMan What I am saying is that I can't patch stuff in the kernel to get this running, you should open an issue at the raspeberrypi linux kernel fork:
https://github.com/raspberrypi/linux

or who ever maintains that driver.

guysoft, 2020-01-15:
Closing if no reply