guysoft/OctoPi #611: [BUG] Problems with timesyncd syncing as expected

OutsourcedGuru, 2019-10-24:
#### What were you doing?
I was attempting to boot my Raspberry Pi 3B computer like I do every morning. It's important to note that this is a problem that's related to both rebooting (long delays at the end) as well as a freeze during the first boot of the day (which is significant).

Three+ minutes on this line as it's shutting down (prior to rebooting):
```
"restore / save the current clock..."
```
Stuck on this line during the first boot of the day (but it's probably the next line that's freezing since this one indicated OK):
```
 [ OK ] Started dphys-swapfile - set up, mount/unmount, and delete a swap file.
```

I note that this occurs more often when I've connected an Ethernet cable as well. [This may be significant.](https://raspberrypi.stackexchange.com/a/49388/97094)

#### What did you expect to happen?
I expected a standard boot to service and I expected that there wouldn't be any unnecessarily-long delays in a reboot.

#### What happened instead?
First boot of the morning seems to require at least one reboot all this week, for example.

#### Did the same happen when running OctoPrint in safe mode?
N/A

#### Version of OctoPi
0.15.1

#### Printer model & used firmware incl. version
Smoothieboard 5xC 1.1, Marlin (but it's not pertinent here)

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

## Troubleshooting
I've literally spent hours chasing this down. It seems like the `fake-hwclock` in Raspbian tries to check-and-save the current time at shutdown. This appears to be the reason for the 3.5-minute delay sometimes.

I ran `date` and note that the time is sometimes as much as 60 seconds off from the time on my MacBook. I ran `sudo timedatectl set-ntp True` thinking that it might be turned off but I don't think that this has changed anything.

The default file seen here is part of the OctoPi image:

`cat /etc/systemd/timesyncd.conf`
```
[Time]
#NTP=
#FallbackNTP=0.debian.pool.ntp.org 1.debian.pool.ntp.org 2.debian.pool.ntp.org 3.debian.pool.ntp.org
#RootDistanceMaxSec=5
#PollIntervalMinSec=32
#PollIntervalMaxSec=2048
```

Does this mean that it's not configured? I did a manual `timedatectl --adjust-system-clock` and the clock appears to be closer to reality. And now, I've just run `date` again and it's slow by 13 minutes. grrr (I'm currently on wi-fi only, for what it's worth.)

I've just now added:

```
/sbin/iptables -t mangle -I POSTROUTING 1 -o wlan0 -p udp --dport 123 -j TOS --set-tos 0x00
```

...before the final line of `/etc/rc.local` and rebooted. It's still about 40 seconds slow right now after the reboot so I don't think that was the problem.

OutsourcedGuru, 2019-10-24:
Also, I saw something somewhere in which—if you're using `ntpd` as a service that you need to add a line in `/etc/rc.conf` which indicates `DAEMONS=(!hwclock ntpd ntpdate)` from [here](https://raspberrypi.stackexchange.com/a/267/97094).

So maybe it's necessary to stop loading the fake version if an actual time-synchronization service is running. Perhaps they're fighting over who sets the time...?

guysoft, 2019-10-29:
Have you found a workaround?
Is this something happening in vanilla OctoPi or did you make changes?

OutsourcedGuru, 2019-10-30:
It's vanilla, dude. I haven't found a work-around yet, to be honest.

Not sure if it's related but 1) the power was out here in NORCAL since Saturday and 2) when I tried to bring things up this morning it probably took ten reboots for it to pull its head out. It then finally worked great all day and without a fuss. (I note that the Ethernet cable is unplugged now; perhaps it's worse with it in but I still need more troubleshooting on that.)

guysoft, 2022-03-20:
@OutsourcedGuru 
So now we have this line in CustomPiOS https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/network/start_chroot_script#L48 
Taken from your comment here: https://github.com/guysoft/OctoPi/issues/611#issue-512176842

However the new Rpi OS does not ship with iptables, it ships only with the ip command.

I could install it on all custompios images, but rather not just because of this command.
Is there any chance to convert it to the ip command? Or could you point me to where you found it so I know better what it does?