guysoft/OctoPi #615: Logitech C270 - Issue with video feed

Jademalo, 2019-11-08:
#### What were you doing?

This happens every time I boot up the raspberry pi, regardless of which USB port the camera is in. Rarely it will act normally, but most of the time this issue happens. I've also tried jiggling the cable in every possible way, it seems to be software.

#### What did you expect to happen?

The webcam to look normal

#### What happened instead?

[image]

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPi

0.17.0

I have read the FAQ.

John-Mc, 2019-11-08:
This is likely not a bug report, since the C270 is one of the more popular cameras out there for use with the Pi and OctoPrint.

What model Pi, what are you using for a power supply, and what other loads do you have plugged in to your Pi? 
Does the camera work well on other computers? Does it work well on a Pi running something other than OctoPi?

Jademalo, 2019-11-08:
I'm using a 2B, and a raspberry pi branded power supply. Only other things plugged into the pi are the printer and a little usb wifi adapter.

What's weird about all of this is it sometimes works absolutely fine. I turned the printer on to check again just now, and it's absolutely perfect.

[image]

I can't test the last two due to the nature of the setup, that would be a huge amount of rewiring. I also don't have a spare SD card to load a normal setup on.

This has only been an issue since I upgraded to the 0.17.0 build, which is why I have a feeling it's software rather than hardware. I never had an issue with this in the past.

guysoft, 2019-11-10:
Solved, closing. not really a bug. ping if anything changes.

closed 2019-11-10 by guysoft

Jademalo, 2019-11-10:
It's not solved, it's intermittent. Simply restarting the Pi can be enough to fix it, but also it can be enough to break it. There's maybe a 1/5 chance of the webcam working correctly.

I'm beginning to think it's got something to do with mjpeg streamer starting too quickly after boot, or not initialising correctly for whatever reason. Is there any way I can add some sort of delay to this? I don't know how OctoPi is configured with regards to loading mjpeg-streamer.

Jademalo, 2019-11-10:
Ok, this is officially extremely weird.

No amount of waiting on the webcamd service will solve this. However, if I restart the service after the first load, it will work every time.

Disabling the service from automatically loading and starting it manually after OctoPrint has fully started still results in the issue, but if I restart the service after this then the webcam works perfectly.

Jademalo, 2019-11-10:
Right, got a workaround. It's an awful workaround, but it has actually fixed my problem.

I added a script to the last line of rc.local to restart the service after a minute, and now every time I reboot the Pi the webcam is working.

```sh
#!/bin/sh
sleep 60
systemctl restart webcamd
```

I have no idea what is causing this in the first place, but this seems to be 100% a bug with the first run of the webcamd daemon after reboot.

guysoft, 2019-11-10:
Looks like something is not powering up in time from the looks of it. I am afraid I have no solution beyond this workaround.

Jademalo, 2019-11-10:
Alright, thanks. At least I've got a workaround.

I'm going to look into making a simple plugin that restarts it once OctoPrint has loaded, that should be a bit more reliable too if it does it once OP is loaded.