guysoft/OctoPi #616: Raspberry Pi 3 B - doesn't detect USB connection with the printer

Tri125, 2019-11-17:
#### What were you doing?

1. Connected my Monoprice mini v2 printer via usb to a freshly installed Octopi
2. Octoprint doesn't list the serial port in the dropdown
3. ``dmesg | tail`` doesn't list any events in regard to USB when I'm connecting the printer

#### What did you expect to happen?
I expected the Pi to detect that a new serial device was connected on the usb port.

#### What happened instead?
The Pi didn't detect the new USB device.

I have tried every ports and different USB cable. When connected to my Windows 10 machine I see the printer listed with the right device name which confirms that both the cable and the printer USB interface works.

When connecting a USB keyboard to the Pi, the device is listed with ``dmesg`` which tells me that the Pi port works.

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPi
0.17.0

#### Printer model & used firmware incl. version
Monoprice Mini v2 (v41.156.2)

#### Screenshot(s)/video(s) showing the problem:
[image]

[image]

I have read the FAQ.

guysoft, 2019-11-17:
Can you get the output of ``lsusb`` or the product+vendor id of the keyboard?
Can you try connecting another keyboard just to see if this is an issue with the particular keyboard or in general?

Tri125, 2019-11-17:
My bad, I should have taken the snapshot with the keyboard connected.

On the first execution only the printer is connected. On the second execution I connected a keyboard. We can see that it is working but not the printer.
[image]

Printer is still connected when I took that screenshot. The keyboard was connected and I unplugged it to take the screenshot. We can see that it correctly detect the keyboard but not the printer.
[image]

Tri125, 2019-11-24:
Small update, I ordered new USB cables just in case and still same problem.
The printer also can't connect to a Pi 3B+ so it seems to be an issue with the distro.

guysoft, 2019-11-24:
Does it connect to vanilla raspbian?

fwdixon, 2020-06-24:
I had similar issue with 0.17.0 connecting with my SKR v1.4 Turbo running Marlin 2.0.5.3. Octopi was mounting the printer as a USB mass storage device.

Downgrading to 0.16.0 fixed the issue.

WhatTakWhat, 2021-01-10:
Same issue, with OctoPi 0.17.0 and with 0.18. + Octoprint 1.5.2 after Marlin 2.0.7 on SKR 1.4 turbo and Raspberry Pi 4 with 2GB. 

I will try OctoPi 0.16. 

By the way, any updates will fix it?

Alpha3D-it, 2023-02-22:
S.O.S Guys
I have the same issue, I can not make Octoprint recognize my printer Ender 3 V2, the raspBerrypi is well recognized on the network but i tried connect the printer with different USB cables(micro Usb), and Octoprint says "No serial Port Found" when I code "sudo dmesg | tail " and " sudo lsusb " I get:
[image]

guysoft, 2023-02-23:
@Alpha3D-it it says there is over-current detected. Somthing is flooding your usb port.

Alpha3D-it, 2023-02-25:
@guysoft 
Thanks for comment.
Wich usb port u mean?, the printer  Micro usb or the computer usb?

guysoft, 2023-02-28:
From the log it looks like the printer USB port. Or if you are plugging something else there