guysoft/OctoPi #618: Plugin Manager broken

gknops, 2019-12-14:
After a vanilla install the Plugin Manager is not working:

`The pip command could not be found or does not work correctly for this installation of OctoPrint - please consult the log file for details and if necessary configure it manually. No installation and uninstallation of plugin packages is possible while pip is unavailable.
`

`It looks like OctoPrint has currently no connection to the internet. Due to that it's not possible to install new plugins from the plugin repository or archive URLs.
`

Internet connectivity is fine. Logfile says:

`2019-12-14 13:55:24,028 - tornado.access - WARNING - 404 GET /static/webassets/packed_plugins.css?53911270 (fe80::c72:befa:86ae:9c05) 5.15ms
2019-12-14 13:55:24,052 - tornado.access - WARNING - 404 GET /static/webassets/packed_plugins.js?e1bede17 (fe80::c72:befa:86ae:9c05) 6.08ms
`

guysoft, 2019-12-15:
Hey, this is an OctoPrint issue, not an OctoPi issue. Moreover if this is a vanilla install its unlikely a bug, please post to the OctoPrint community to sovle your issue: https://community.octoprint.org/

closed 2019-12-15 by guysoft

gknops, 2019-12-15:
Not sure about that. The issue is about `PATH`, and the fix (at least for me) was to change it in `/etc/init.d/octoprint` like so:

`	PATH=/usr/local/sbin:/usr/local/bin:/sbin:/bin:/usr/sbin:/usr/bin:/home/pi/oprint/bin
`

guysoft, 2019-12-15:
That's by design. We don't want the virtualenv default in the distro. Its a virtualenv in order to stay separate

gknops, 2019-12-15:
So where would be the appropriate place to set it? Setting PATH certainly feels platform specific to me and therefore should be part of OctoPi? Not trying to be difficult, but the current OctoPi image results in a broken plugin manager. Worse the error messages will send the user on wild goose chases (pip corrupt or not found, internet connection down). Given that OctoPi tries to make things easy for Pi users, I think it should address this.