guysoft/OctoPi #621: Limited camera resolution on Raspberry Pi 4 with VL805 firmware 0x000137ab

szopen111, 2019-12-22:
#### What were you doing?
Trying to watch the webcam stream on Raspberry Pi 4 with VL805 firmware 0x000137ab.
Webcam: "ID 1871:0141 Aveo Technology Corp. Camera"
It worked before. Now it doesn't work (I can see only "webcam stream loading")
(**for a possible solution see below**)

#### What did you expect to happen?
Webcam stream displayed.

#### What happened instead?
No webcam stream displayed. I can see only "webcam stream loading"

#### Did the same happen when running OctoPrint in safe mode?
yes

#### Version of OctoPi
OctoPrint 1.3.12 running on OctoPi 0.17.0

#### Printer model & used firmware incl. version
n/a

#### Screenshot(s)/video(s) showing the problem:
n/a

#### Investigation
The problem seems to be connected to a bug in `0x000137ab` firmware of `VL805`.

First enable `uvcvideo` debugging and show the log:
``` bash
echo 0xffff > /sys/module/uvcvideo/parameters/trace
dmesg -w
```

Streaming very low resolutions seems to work fine: 
``` bash
./mjpg_streamer -o "output_http.so -w ./www-octopi -n" -i "input_uvc.so -y -r QVGA -d /dev/video0"
```

But trying to stream even in `640x480` doesn't work:
``` bash
./mjpg_streamer -o "output_http.so -w ./www-octopi -n" -i "input_uvc.so -y -r VGA -d /dev/video0"
```
you can see a lot of `uvcvideo: USB isochronous frame lost (-18).` lines in the `dmesg` logs and `i: select() timeout` in the `mjpg_streamer`'s output.

Now run (as root):
``` bash
setpci -s 01:00.0 0xD4.B=0x41
```
(see: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=244421 for details)

Problem solved, the stream works fine!

#### Possible solution
You have to run the `setpci` command after each boot, so I've modified `/root/bin/webcamd` to do it for me:
``` diff
--- webcamd.orig        2019-12-22 22:42:23.812714213 +0100
+++ webcamd     2019-12-22 22:51:21.563037412 +0100
@@ -116,6 +116,33 @@
 function runMjpgStreamer {
     input=$1

+    # There are problems with 0x000137ab firmware on VL805.
+    # It seems to work fine on 0x00013701
+    # It may be upgraded someday... display a warning and do nothing (it may just work!)
+    echo "Checking for vl805..."
+    if [[ -f /usr/bin/vl805 ]]; then
+        VL805_VERSION=$(/usr/bin/vl805)
+        VL805_VERSION=${VL805_VERSION#*: }
+        echo " - version 0x${VL805_VERSION} detected"
+        case "$VL805_VERSION" in
+            00013701)
+                echo "    - nothing to be done. It shouldn't cause USB problems."
+                ;;
+            000137ab)
+                echo -e "    - \e[31mThis version is known to cause problems with USB cameras.\e[39m"
+                echo -e "      You may want to downgrade to 0x0013701."
+                echo -e "    - [FIXING] Trying the setpci -s 01:00.0 0xD4.B=0x41 hack to mitigate the issue"
+                echo -e "      It disables ASPM L1 on the VL805"
+                setpci -s 01:00.0 0xD4.B=0x41
+                ;;
+            *)
+                echo "   - unknown firmware version. Doing nothing."
+                ;;
+        esac
+    else
+        echo "  - It seems that you don't have VL805 (Raspberry Pi 4). There should be no problems with USB (a.k.a. select() timeout)"
+    fi
+
     pushd $MJPGSTREAMER_HOME > /dev/null 2>&1
         echo Running ./mjpg_streamer -o "output_http.so -w $camera_http_webroot $camera_http_options" -i "$input"
         LD_LIBRARY_PATH=. ./mjpg_streamer -o "output_http.so -w $camera_http_webroot $camera_http_options" -i "$input" &
```
I'm not an expert on Linux. The real cause and/or solution may be completely different.

guysoft, 2019-12-30:
Would love a PR if you have a fully drafted solution.

guysoft, 2019-12-31:
Fixed by #623

closed 2019-12-31 by guysoft

looxonline, 2020-01-20:
You should also add FW version 000137a8. Just spent half a day struggling to figure it out only to find that I had that version out the box and it suffers from the same issue. In any case I'm using a PS3eye which does not support mjpg and has to encode on the fly so I changed over to ustreamer which uses 9% CPU instead of 35% with mjpg. Still suffered with the same issue though. PCIe fix works.