guysoft/OctoPi #625: Several issues since flashing Octo 0.17.0 with Octoprint

NovaViper, 2020-01-11:
#### What were you doing?
I just recently had to reflash my pi and now I'm having *several* issues.
1. My printer now is constantly pausing, like not actually pausing the print, but the printer freezes in place for a split second, then it moves, then repeats. When I print from the printer's sd card, it never do this
2. The image *did not* come with pip, so I am unable to install plugins.
3. I can no longer flash with the plugin FirmwareUpdater because the Raspbian image that 0.17.0 changes up how the pi reads USBs, see [here](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/112)

1. Flash OctoPi 0.17.0
2. Setup Octoprint like normal
3. For the first issue, when trying to go and install the plugins, you will immediately see that pip is not included

#### What did you expect to happen?
None of these issues to occur

#### What happened instead?
These issues occured

#### Did the same happen when running OctoPrint in safe mode?
For the first and third issue, I had no plugins installed at the tip (so safe mode made no difference). For the second, I haven't gotten the chance to test it in safe mode

#### Version of OctoPi
0.17.0

#### Printer model & used firmware incl. version
Creality Ender 3 with SKR V1.3 board, Marlin 2.0

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

NovaViper, 2020-01-11:
I can confirm going back to OctoPi 0.16.0 resolves all of the following issues stated

guysoft, 2020-01-12:
Hey, I am not sure this is an OctoPi issue yet. You might have more luck for people understanding your issue at: http://discourse.octoprint.org/ . I warmly suggest you post there, along with your ``octoprint.log``, otherwise there is no way to debug this without the hardware.

The image does come with pip, which is located in a virtual env at: ``/home/pi/oprint/bin/pip``

NovaViper, 2020-01-13:
@guysoft This is the troubling thing though, there are no logs that show any sort of error (nor do I have access to them anymore since I went back to OctoPi 0.16.0) that would explain what's thoroughly going on. And the thing with pip is that I event SSHed into the pi and tried to find pip, but it was no where to be found on it whatsoever. All of the issues all seem to stem from the actual Octo image, not Octoprint. I went and reflashed OctoPi 0.16.0 and even updated everything to match what OctoPi 0.17.0 has, and yet I didn't experience any of the issues I stated already while doing this. 

I do have some logs from when I was trying to find pip, if that helps any.
[octoprint.log](https://github.com/guysoft/OctoPi/files/4051899/octoprint.log)
[octoprint (1).log](https://github.com/guysoft/OctoPi/files/4051900/octoprint.1.log)

guysoft, 2020-03-01:
Ok, did you post this where I told you at  http://discourse.octoprint.org/  ?

NovaViper, 2020-03-05:
@guysoft Hey, sorry for the late reply! I did go and post the problem on the forums [here](https://community.octoprint.org/t/several-issues-since-flashing-octo-0-17-0-with-octoprint/16679/5). I have to go and test out what foosel said to try but I haven't gotten the chance to since my courses have ramped up

NovaViper, 2020-06-01:
This tip that fossel mentioned fixed the issue for me: https://community.octoprint.org/t/several-issues-since-flashing-octo-0-17-0-with-octoprint/16679/5?u=novaviper

closed 2020-06-01 by NovaViper