guysoft/OctoPi #626: USB webcam not working on startup

ScruffR, 2020-01-14:
#### What were you doing?

After a SD card crash I finally got round to upgrading to the lates release but noticed that my Logitech Pro 9000 didn't work.

#### What did you expect to happen?

Of course I expected the webcam stream to show

#### What happened instead?

but it only ever keeps displaying this
[image]
for hours on end

#### Did the same happen when running OctoPrint in safe mode?

Yes, just the same.

#### Version of OctoPi

I'm running 
[image]
on a RPi 3B+

#### Printer model & used firmware incl. version

although irrelevant for the issue at hand: Creality CR-10

#### Screenshot(s)/video(s) showing the problem:

see _"What happened instead?"_

I have read the FAQ.

However, I found that doing this resolves the issue till next reboot 
```
sudo systemctl restart webcamd
```
After boot `systemctl status webcamd` reports this
```
● webcamd.service - the OctoPi webcam daemon with the user specified config
   Loaded: loaded (/etc/systemd/system/webcamd.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2020-01-13 04:30:48 GMT; 1 day 4h ago
 Main PID: 440 (mjpg_streamer)
    Tasks: 5 (limit: 2077)
   Memory: 3.7M
   CGroup: /system.slice/webcamd.service
           └─440 ./mjpg_streamer -o output_http.so -w ./www-octopi -n -i input_uvc.so -r 640x480 -f 10 -d /dev/video0

Jan 13 04:30:57 octopi mjpg_streamer[440]: MJPG-streamer [440]: HTTP TCP port........: 8080
Jan 13 04:30:57 octopi mjpg_streamer[440]: MJPG-streamer [440]: HTTP Listen Address..: (null)
Jan 13 04:30:57 octopi mjpg_streamer[440]: MJPG-streamer [440]: username:password....: disabled
Jan 13 04:30:57 octopi mjpg_streamer[440]: MJPG-streamer [440]: commands.............: disabled
Jan 13 04:30:57 octopi mjpg_streamer[440]: MJPG-streamer [440]: starting input plugin input_uvc.so
Jan 13 04:30:57 octopi mjpg_streamer[440]: MJPG-streamer [440]: starting output plugin: output_http.so (ID: 00)
Jan 13 04:31:02 octopi webcamd[345]:  i: select() timeout
Jan 13 04:31:02 octopi webcamd[345]:  i: cleaning up resources allocated by input thread
Jan 13 04:31:02 octopi mjpg_streamer[440]: MJPG-streamer [440]: select() timeout
Jan 13 04:31:02 octopi mjpg_streamer[440]: MJPG-streamer [440]: cleaning up resources allocated by input thread
```
and after the restart this
```
● webcamd.service - the OctoPi webcam daemon with the user specified config
   Loaded: loaded (/etc/systemd/system/webcamd.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-01-14 09:20:51 GMT; 19s ago
  Process: 4148 ExecStart=/root/bin/webcamd (code=exited, status=0/SUCCESS)
 Main PID: 4173 (mjpg_streamer)
    Tasks: 4 (limit: 2077)
   Memory: 1.3M
   CGroup: /system.slice/webcamd.service
           └─4173 ./mjpg_streamer -o output_http.so -w ./www-octopi -n -i input_uvc.so -r 640x480 -f 10 -d /dev/video0

Jan 14 09:20:50 octopi mjpg_streamer[4173]: MJPG-streamer [4173]: HTTP Listen Address..: (null)
Jan 14 09:20:50 octopi webcamd[4148]:  o: username:password....: disabled
Jan 14 09:20:50 octopi mjpg_streamer[4173]: MJPG-streamer [4173]: username:password....: disabled
Jan 14 09:20:50 octopi webcamd[4148]:  o: commands.............: disabled
Jan 14 09:20:50 octopi mjpg_streamer[4173]: MJPG-streamer [4173]: commands.............: disabled
Jan 14 09:20:50 octopi mjpg_streamer[4173]: MJPG-streamer [4173]: starting input plugin input_uvc.so
Jan 14 09:20:50 octopi mjpg_streamer[4173]: MJPG-streamer [4173]: starting output plugin: output_http.so (ID: 00)
Jan 14 09:20:51 octopi webcamd[4148]: Done bring up all configured video device
Jan 14 09:20:51 octopi webcamd[4148]: Goodbye...
Jan 14 09:20:51 octopi systemd[1]: Started the OctoPi webcam daemon with the user specified config.
```
After that the stream works even after a restarting OctoPrint - however when rebooting the RPi the problem reoccures.

Could a delayed start of `webcamd` be a possible solution?

tablatronix, 2020-09-02:
Same issue, stopped working a while ago, not sure why, just got around to debugging and  tried this and it works.

dragonfly4, 2020-12-21:
Same issue on 0.18 RC1
Spent 2 hours for nothing so thanks for the workaround :-)

derkork, 2021-08-04:
Running OctoPi 0.18.0 / OctoPrint 1.6.1, having the same issue with a Logitech QuickCam Pro 9000. Restarting webcamd fixes it here as well.

GitFunkee, 2022-07-27:
Brilliant! The restart fixed my issue as well.