guysoft/OctoPi #627: Unable to build image since Setuptools v45.0.0 release

angudu, 2020-01-15:
I'm having troubles for build an image since some days ago.
It seams that Setuptools v45.0.0 breaks virtualenv on python2. [https://github.com/pypa/virtualenv/issues/1493](url)

The build.log generated:
[build.log](https://github.com/guysoft/OctoPi/files/4065642/build.log)

Any fix for prevent use Setuptools v45.0.0?

angudu, 2020-01-15:
Replacing this code:
sudo -u pi virtualenv oprint
sudo -u pi /home/pi/oprint/bin/pip install --upgrade pip

By this code:
sudo -u pi virtualenv --no-setuptools oprint
sudo -u pi /home/pi/oprint/bin/pip install --upgrade pip
sudo -u pi /home/pi/oprint/bin/pip install "setuptools<45"

On file:
OctoPi/src/modules/octopi/start_chroot_script

Seems to solve the issue temporarily.

guysoft, 2020-01-15:
Related to https://github.com/pypa/pipenv/issues/4093

I am not sure how they solved it, something about ``index-url``

What do you mean by "Seems to solve the issue temporarily"? Does it mean it works?
Would you like to PR that? Or should I do it?

angudu, 2020-01-15:
**Does it mean it works?**
Yes, the image is build succeeded.
[build.log](https://github.com/guysoft/OctoPi/files/4067178/build.log)

**What do you mean by "Seems to solve the issue temporarily"?**
Because I hope this issue will be fixed soon on virtualenv.

**Would you like to PR that? Or should I do it?**
If you can do it, I would prefer it. Thank you.

guysoft, 2020-01-15:
Ok testing it here

commit a3c7da1f (2020-01-16): Fixes #627
closed 2020-01-16 by guysoft