guysoft/OctoPi #630: sd card full of timelapses prevents server from running

thoughtcastapp, 2020-01-27:
Per this thread:

https://community.octoprint.org/t/octopi-connects-to-my-network-but-shows-the-server-is-down/15641/2

My sd card filled with timelapses, and now my server won't start...don't know why, might be worth setting octopi up so that when the card is full it stops recording timelapses instead of locking someone out (and now im struggling to find the files and delete them via ssh)...

tedder, 2020-01-27:
This is how most operating systems work; if anything, Octoprint's timelapse.py might stop recording timelapses when the disk is getting full (so this isn't an octopi problem).

On the other hand we'd start seeing support requests for "timelapses aren't recording?!!!!".

thoughtcastapp, 2020-01-27:
all i know is my octopi wont boot, and the only thing i know might be causing it is the overload of timelapses...and on top of that, i cannot find them for the life of me (either via ssh or with the sd card plugged into my mac)...

thoughtcastapp, 2020-01-28:
i just erased my sd card and reinstalled the software, problem solved

closed 2020-01-28 by thoughtcastapp

guysoft, 2020-01-28:
Duplicate of: https://github.com/guysoft/OctoPi/issues/354