guysoft/OctoPi #633: [Request] Stop RPi backpowering printers

qwert95, 2020-02-26:
My printer is a just bought Ender 3.
Apparently it's known in the web : Raspberry Pi boards can (and my RPi 3 does) backpower the printer through the USB port, to the point that if I turn the printer off from the power supply, the little display will stay on.
This was proven to be potentially harmful to multiple components of the printer.

I'm here to kindly ask if someone with more time than me can implement what I found [here](https://www.reddit.com/r/3Dprinting/comments/8ng5z5/is_providing_power_from_rpi_to_ender_3_board/e79hvdn/) as a possible fix to this.

Afaik in USB cables power is separated by data, so probably what we need is to simply disconnect the power from the printer USB port during the rpi boot.

If anybody is willing to do it but doesn't have an Ender 3, I'm of course available to test it.

Thank you.

OricTosh, 2020-02-26:
Rather than implement a software program to monitor/control the USB +5v line, it would be better to cut the USB cable and rejoin the 2 data cables and not the 5V line. 
USB is a 4 wire connection,here some info on USB cables 
http://www.hobbytronics.co.uk/usb-connector-pinout 
I had a similar problem with a 5V powered USB hub with keyboard plugged into the hub pulling 250mA which stopped the pi from booting, it was the pi 1 (had problems with the current limiters 100mA) so I cut the red wire in the cable, it worked fine.
The printer shouldn't need the 5V volts from the USB host computer as the printer's board will drop the voltage to 5V from the PSU.
Quite happy to supply a cable if needed, thou shouldn't be that hard to mod an existing cable, just cut the red wire and insulate the ends to avoid shorting.

guysoft, 2020-02-27:
Cool, I can ship it in OctoPi, but I will need someone to write the plugin. 
Also note this only works on some USB hubs. There is a list.
I would only do this though if someone shows interest in helping. I don't really have the hardware to test this.

NameOfTheDragon, 2020-04-21:
While this isn't strictly an OctoPrint/OctoPi problem, it's not ideal to require users to modify their hardware in order to have the software work. It would be great to nudge users into the "pit of success" as far as reasonably possible. A software solution in the Raspberry Pi would be very helpful.

Unfortunately I can't help with the development because I have no relevant experience but I would be very happy to test using my Pi 3B+ and Ender-3. I'm using an after-market main board in my Ender-3 but that shouldn't make any difference.

touche112, 2020-04-22:
This is already a plugin: https://github.com/OutsourcedGuru/OctoPrint-USBControl

Note that this repo hasn't been updated in a while and doesn't have Pi4B support yet.  Hopefully I'll be able to get it up to date and pull req it soon.

guysoft, 2020-04-26:
@OutsourcedGuru hangs around here, would value input if you have any suggestions

touche112, 2020-04-26:
> @OutsourcedGuru hangs around here, would value input if you have any suggestions

I've put in a pull req on his repo updating for RPi4 support. Should be good to go for pretty much everyone after it's merged.