guysoft/OctoPi #635: /boot/octopi-wpa-supplicant.txt and /etc/wpa_supplicant/wpa_supplicant.conf

yet-another-average-joe, 2020-03-29:
#### What were you doing?

I edit /boot/octopi-wpa-supplicant.txt : SSID and password

#### What did you expect to happen?

I expect new SSID and password to be taken into account

#### What happened instead?

SSID and password are not updated

#### Did the same happen when running OctoPrint in safe mode?

Not tested... (not relevant)

#### Version of OctoPi

0.17.0

#### Printer model & used firmware incl. version

not relevant

#### Screenshot(s)/video(s) showing the problem:

not relevant

#### workaround

/etc/wpa_supplicant/wpa_supplicant.conf has to be edited by hand. This is not user friendly.
With previous OctoPi release, /etc/wpa_supplicant/wpa_supplicant.conf was a shortcut to /boot/octopi-wpa-supplicant, and therefore any change in the "boot" file was taken into account. (in Pixel file explorer, wpa_supplicant.conf icon shows the shortcut arrow, not in OctoPi 0.17.0.

Pleeeeaaaaaseeee, give us back the shortcut ! It is so easy to edit a file in the boot partirtion, just inserting the SD in a computer ! Also, it seems to me that editing WiFi parameters in raspi-config has no effect. This is a bit confusing.

jsfour, 2020-03-29:
Im seeing this issue as well.

guysoft, 2020-03-29:
Strange,  it should be a shortcut, is this with a fresh new installed of the current stable release?

jsfour, 2020-03-29:
Im using the download link in the readme. Is that the stable release?

guysoft, 2020-03-31:
> Im using the download link in the readme. Is that the stable release?

Yes, that should be right, its really strange, I will have to take a look

yet-another-average-joe, 2020-03-31:
On my side I will reinstall OctoPi, then Pixel, and see what happens.

I have to say I run the RasPi with a USB WiFi dongle, the onboard one being deactivated. wpa_supplicant.conf was edited by hand, but opening a shortcut and saving it should not turn it into a plain text file AFAIK...

yet-another-average-joe, 2020-04-01:
I reproduced all steps (a lot !), including enabling a WiFi dongle / disabling onboard WiFi. Nowhere the symbolic link was replaced with a text configuration file. The 'l' attribute is still there, and the color is still cyan.

I also played with gedit...

No problem. Not even the smallest one.

No idea how, where, when the problem occured. No idea how @jsfour had the same issue.

The only explanation I could think of is overwriting the etc shortcut file with the boot one at some point, renaming it, and not rembering. But why would we do this ?

This seems to be a non issue, and the result of "some" mistake by "some" users.

I'm not a linux specialist. But I found how to create a symbolic link using Google. If this issue occurs again, it is very easy to solve.

@guysoft : I don't close this issue (who knows), but, again, IMHO the problem is most likely somewhere between the seat and the Raspi...

guysoft, 2020-04-02:
I have another guess. There is another way for Raspbian to set the wifi settings. [Here is a  link to it (DONT USE THIS)](https://www.raspberrypi.org/documentation/configuration/wireless/headless.md). It involves putting a ``wpa_supplicant.conf`` file in /boot. which is then copied, presumably over the existing symbolic link. If you used that, it would very likely break out method.

If @jsfour has done it, or you , @yet-another-average-joe know of it, it would be a good indicator that this is the case and I can close the issue.

yet-another-average-joe, 2020-04-02:
yes ! I close the issue. Whatever was the mistake, it is easy to recreate a symbolic link, using a console or Pixel.

closed 2020-04-02 by yet-another-average-joe