guysoft/OctoPi #639: Pi User Keeps Losing Sudo

Kooftness, 2020-04-19:
Octoprint 1.4.0
Octopi 0.17.0

I have re-flashed the image 3 times. for some reason the first day that I am using it to configure everything in the terminal the pi user has sudo privileges. but if I leave it overnight or for a few days come back to the terminal the Pi user has lost sudo privileges. everything I try to sudo I get 

Sorry, user pi is not allowed to execute '/usr/bin/raspi-config' as root on octopi.

tedder, 2020-04-20:
Can you give more text? e.g., what is the exact command you are typing, what is the exact error? Are you asked for "password for pi" initially, and are you asked for it again the next day? You can copy/paste the text or use a screenshot.

Based on "user pi is not allowed" I suspect your command is incorrect.

Kooftness, 2020-04-20:
[image]

tedder, 2020-04-20:
That is interesting- it's not what I expected. But **your Pi is accessible from the public internet**. It's likely it "loses sudo" because it's being hacked. In fact, I was able to ssh to it with the default password, though I got a special warning that Raspbian (or octopi?) is configured to give to external hosts.

Again, **you are getting hacked**.

Kooftness, 2020-04-20:
i am not sure how that is possible. i connect to the web by octopi.local i have not set any port forwarding or anything as i am not sure how to. I would like to find a simple way to access the pi remotely but until now everything has been local

guysoft, 2020-04-21:
Octopi has the ``password-for-sudo`` module enabled, [which runs this](https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/password-for-sudo/start_chroot_script)
So 0.17 should not have passwordless sudo. However it looks like the password was changed.

tedder, 2020-04-21:
For fun, @Kooftness, can you paste the results of this command?
```
ls -l /etc/passwd /etc/sudoers /etc/sudoers.d/
```

It's definitely accessible from the internet. Your router is configured in a bad way.