guysoft/OctoPi #642: TL-wn725n not working with octopi 0.17.0

johannesnicolaus, 2020-04-23:
#### What were you doing?

Raspberry pi model: raspberry pi 1 model b+
WIFI dongle: TL-wn725n V1

1. Flashed Octopi 0.17.0 from previously flashed 0.16.0
2. Setup `octopi-wpa-supplicant.txt` the same way I did when I used 0.16.0
3. Turned on raspberry PI 
4. Octopi would not connect to wifi

I also tried installing the following driver 8188eu-4.19.75-v8-1270.tar.gz from http://downloads.fars-robotics.net/wifi-drivers/8188eu-drivers/ but did not work.

Now I moved back to 0.16.0 and it works without any problem.

#### What did you expect to happen?
Octopi to connect to wifi

#### What happened instead?
Octopi did not connect to wifi

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPi
0.17.0

I have read the FAQ.

guysoft, 2020-04-25:
Please provide information about your chipset.
Provide the output of:
```
lsusb -v
```
Note I am not sure this is an OctoPi issue, since we are based on Raspbian.
Basing on source: https://www.raspberrypi.org/forums/viewtopic.php?t=251182

johannesnicolaus, 2020-04-26:
Here is the output of `lsusb -v`

```
Bus 001 Device 004: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x8179 RTL8188EUS 802.11n Wireless Network Adapter   
  bcdDevice            0.00
  iManufacturer           1 Realtek
  iProduct                2 802.11n NIC
  iSerial                 3 00E04C0001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```

I'm not sure too, but octopi 0.16.0 worked fine so I thought it might have something to do with octopi 0.17.0. For now, I am using octopi 0.16.0 without any problem.

guysoft, 2020-05-05:
Its a Raspbian issue, see: https://www.raspberrypi.org/forums/viewtopic.php?t=246587

There is an issue with workaround solutions here: https://github.com/raspberrypi/linux/issues/3102

closed 2020-05-05 by guysoft