guysoft/OctoPi #648: [cosmetic] no clm_blob available (err=-2), device may have limited channels available

burnbabyburn, 2020-05-12:
#### What were you doing?

1. Boot RPI3 with fresh Octoprint
2. Enter dmesg

#### What did you expect to happen?
dmesg output like:
```
[    5.192799] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    5.461112] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    5.475481] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Feb 16 2020 22:39:24 version 7.45.98.97 (r724416 CY) FWID 01-bf41ed64
```
#### What happened instead?
dmesg output like:
```
[    5.449096] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    5.449230] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    5.450047] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f
```
#### Did the same happen when running OctoPrint in safe mode?
OS related

#### Version of OctoPrint
Version1.4.0/Version 0.17.0, running on Raspberry Pi 3 Model B Rev 1.2

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log
[iw list](https://gist.githubusercontent.com/burnbabyburn/01237f7cc2c5182aeebf783a5ef47a3d/raw/92cca6a1c3f490728b595e7a1c76857a589dd94a/gistfile1.txt)

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

---
Its just a coesmetic error, since wifi capability does not change. iw list outpts the same. It may break wifi, i've never configured it. But it can not hurt to fix error and update 3 year old firmware.

[CypressFirmwareDL](https://community.cypress.com/docs/DOC-19375)
or
https://github.com/ccrisan/thingos/issues/31

Just replace or add the firmwareblobs from the links above to `/lib/firmware/brcm/brcmfmac43455-sdio*` . Keep brcmfmac43455-sdio.txt

Feel free to close if it's a won't fix. I didn't found any tickets regarding "clm_blob"

guysoft, 2020-05-13:
There is an issue reported in the RaspberryPi kernel with this error: https://github.com/raspberrypi/linux/issues/3415

burnbabyburn, 2020-05-13:
Sorry for interupting your work than. :)
I thought this was patched ages ago in mainline as i found solutions.
The problem you linked is probably related to this issue.
AFAIK/AFAIR - On a PI3B+ this file is neccessary for channel width/ht capabilies

closed 2020-05-13 by burnbabyburn