guysoft/OctoPi #649: [Request] I'm going to add H264/HLS support.

chudsaviet, 2020-05-18:
Hi, @guysoft , @foosel .
I have a working prototype of HLS streaming from RaspberryPi, through not embedded it to OctoPrint UI yet.
However, it's working on a separate web page in all browsers, and provides side MJPG frames for timelapses.

It have couple disadvantages:
1) By design, HLS have lag. I was able to bring it down to 3 - 5 sec.
2) I was not able to make it work with MJPG-streamer at the same time due to V4L2 limitations - one consumer per device.

But it have advantages:
1) H264, which is a effective video codec.
2) Uses Raspberry Pi hardware H264 encoder.
3) Uses standard HTML5 video tag and HLS.js lib, which is well supported.
4) Able to add audio.
5) Able to carry multiple resolutions and bitrates.
6) Future-proof. We will be able to switch to HEVC or AV1 when they become supported.

I just wanted your opinion on if the 3-5 sec lag will be acceptable.

tedder, 2020-05-18:
this is great! Are you using [the new picamera interface](https://picamera.readthedocs.io/en/release-1.13/)?

chudsaviet, 2020-05-19:
@tedder , nope.
Just V4L2 + ffmpeg. It's 100% enough and universal for both USB and Rpi cameras.

guysoft, 2020-05-20:
I think it should be OK, I don't mind shipping both streamers in the official image with the option to try both, let the users decide. We could not ship it as the default in the first release, and see how the demand goes

Also linking related issues  #545 and #529 . Although its been a while, perhaps users there might be eager to test a new streamer.

chudsaviet, 2020-05-20:
Of course, this was my plan.

chudsaviet, 2020-05-23:
Please, look at https://github.com/guysoft/OctoPi/pull/650 .

guysoft, 2021-09-19:
Can this ve closed? There is a source merged,  or is more work needed?