guysoft/OctoPi #652: [BUG] Problems with 'Build OctoPi From within OctoPi '

jiantaolovebingbing, 2020-06-07:
#### What were you doing?

1. Just follow the readme file the 'Build OctoPi From within OctoPi / Raspbian / Debian / Ubuntu' part
2. in last step, it reports an error

#### What did you expect to happen?
The build.log file is  showed as below .
[build1.log](https://github.com/guysoft/OctoPi/files/4741339/build1.log)

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPi
OctoPi 0.17.0
pi 3 model B+

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

[Uploading build1.log…]()

I have read the FAQ.

guysoft, 2020-06-07:
Entry from build.log:
```
losetup: /dev/loop0: detach failed: No such device or address
```
Duplicate of https://github.com/guysoft/CustomPiOS/issues/55

Try running again. The issue is a bug in Docker (which its good you are using) where it does not update in time the loopback devices. It tends to fail once, then work on the next run.

closed 2020-06-07 by guysoft

jiantaolovebingbing, 2020-06-08:
sorry, upload wrong file.
[buildPi.log](https://github.com/guysoft/OctoPi/files/4743778/buildPi.log)

guysoft, 2020-06-09:
This issue has been opened at: #653 Please comment there