guysoft/OctoPi #653: [BUG] Problems with 'build failed, unmounting image...'

jiantaolovebingbing, 2020-06-08:
#### What were you doing?

1. Just follow the readme file the 'Build OctoPi From within OctoPi / Raspbian / Debian / Ubuntu' part
2. in last step, it reports an error

#### What did you expect to happen?
Sorry for that uploading wrong file!
The build.log file is  showed as below .
[buildPi.log](https://github.com/guysoft/OctoPi/files/4745751/buildPi.log)

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPi
OctoPi 0.17.0
pi 3 model B+

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

jiantaolovebingbing, 2020-06-09:
I can find where went wrong, but I don’t know how to fix it.
[image]
thanks

guysoft, 2020-06-09:
Hey, if you look in the logs you can see this is the error, the unmount only happends because the system is clearing up after itself:
```
+ sudo -u pi virtualenv --no-setuptools oprint
sudo: unable to resolve host octopi: Name or service not known
New python executable in /home/pi/oprint/bin/python2
Also creating executable in /home/pi/oprint/bin/python
Installing pip, wheel...
  Complete output from command /home/pi/oprint/bin/python2 - pip wheel:
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Looking in links: /usr/lib/python3/dist-packages, /usr/share/python-wheels/
Collecting pip
  Downloading https://files.pythonhosted.org/packages/43/84/23ed6a1796480a6f1a2d38f2802901d078266bda38388954d01d3f2e821d/pip-20.1.1-py2.py3-none-any.whl (1.5MB)
Exception:
Traceback (most recent call last):
  File "/usr/share/python-wheels/pip-18.1-py2.py3-none-any.whl/pip/_internal/cli/base_command.py", line 143, in main
    status = self.run(options, args)
  File "/usr/share/python-wheels/pip-18.1-py2.py3-none-any.whl/pip/_internal/commands/install.py", line 338, in run
    resolver.resolve(requirement_set)
  File "/usr/share/python-wheels/pip-18.1-py2.py3-none-any.whl/pip/_internal/resolve.py", line 102, in resolve
    self._resolve_one(requirement_set, req)
  File "/usr/share/python-wheels/pip-18.1-py2.py3-none-any.whl/pip/_internal/resolve.py", line 256, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/usr/share/python-wheels/pip-18.1-py2.py3-none-any.whl/pip/_internal/resolve.py", line 209, in _get_abstract_dist_for
    self.require_hashes
  File "/usr/share/python-wheels/pip-18.1-py2.py3-none-any.whl/pip/_internal/operations/prepare.py", line 283, in prepare_linked_requirement
    progress_bar=self.progress_bar
  File "/usr/share/python-wheels/pip-18.1-py2.py3-none-any.whl/pip/_internal/download.py", line 836, in unpack_url
    progress_bar=progress_bar
  File "/usr/share/python-wheels/pip-18.1-py2.py3-none-any.whl/pip/_internal/download.py", line 673, in unpack_http_url
    progress_bar)
  File "/usr/share/python-wheels/pip-18.1-py2.py3-none-any.whl/pip/_internal/download.py", line 897, in _download_http_url
    _download_url(resp, link, content_file, hashes, progress_bar)
  File "/usr/share/python-wheels/pip-18.1-py2.py3-none-any.whl/pip/_internal/download.py", line 617, in _download_url
    hashes.check_against_chunks(downloaded_chunks)
  File "/usr/share/python-wheels/pip-18.1-py2.py3-none-any.whl/pip/_internal/utils/hashes.py", line 48, in check_against_chunks
    for chunk in chunks:
  File "/usr/share/python-wheels/pip-18.1-py2.py3-none-any.whl/pip/_internal/download.py", line 585, in written_chunks
    for chunk in chunks:
  File "/usr/share/python-wheels/pip-18.1-py2.py3-none-any.whl/pip/_internal/utils/ui.py", line 159, in iter
    for x in it:
  File "/usr/share/python-wheels/pip-18.1-py2.py3-none-any.whl/pip/_internal/download.py", line 574, in resp_read
    decode_content=False):
  File "/home/pi/oprint/share/python-wheels/urllib3-1.24.1-py2.py3-none-any.whl/urllib3/response.py", line 494, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/home/pi/oprint/share/python-wheels/urllib3-1.24.1-py2.py3-none-any.whl/urllib3/response.py", line 459, in read
    raise IncompleteRead(self._fp_bytes_read, self.length_remaining)
  File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
    self.gen.throw(type, value, traceback)
  File "/home/pi/oprint/share/python-wheels/urllib3-1.24.1-py2.py3-none-any.whl/urllib3/response.py", line 374, in _error_catcher
    raise ReadTimeoutError(self._pool, None, 'Read timed out.')
ReadTimeoutError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Read timed out.
----------------------------------------
...Installing pip, wheel...done.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 2379, in <module>
    main()
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 724, in main
    symlink=options.symlink)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 996, in create_environment
    download=download,
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 926, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 817, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command /home/pi/oprint/bin/python2 - pip wheel failed with error code 2
Running virtualenv with interpreter /usr/bin/python2
```

The code files when installing pip inside the virtual env.

In particular it fails because of this:

```
ReadTimeoutError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Read timed out.
```
The error ``Read timed out`` means the image is unable to connect to the server. For some reason.

Make sure your machine can reach the host ``files.pythonhosted.org``. If it can there might be an issue in your environment, in that case I would recommend trying the docker build method. Because it makes sure your env works.
If it doesn't then your build machine has connection issues.

jiantaolovebingbing, 2020-06-10:
thanks for the reply!
one more thing, if i want to build my nodejs project inside octopi, i just change the Dockerfile and it 
build success. But I entered the pi system and did not start the nodejs service. 
is there any document or tips could help me to fix this job?
thanks again !

below is my build.log file and Dockerfile file.

[Dockerfile.txt](https://github.com/guysoft/OctoPi/files/4757918/Dockerfile.txt)

jiantaolovebingbing, 2020-06-10:
[successLu.log](https://github.com/guysoft/OctoPi/files/4757924/successLu.log)

jiantaolovebingbing, 2020-06-11:
i thought  what  should be done are below:
1. copy the `build_dist` and `config` file to nodejs project
2.copy the `modules` directory to the nodejs project
3. change the content in `modules/src/octopi/start_chroot_script`. but how to change it ?

closed 2020-06-15 by jiantaolovebingbing

guysoft, 2020-06-15:
Hey @jiantaolovebingbing DId you solve you issue?
Would be great if you share for others how you solve your nodejs issue and how to place the modules.

jiantaolovebingbing, 2020-06-28:
I found a related issue in the issue, but i still not solve it completely.

guysoft, 2020-06-28:
I am not sure how those issues are connected, and I am not following what you are trying to do now because you have 2-3 issues mixed in to one - it would be nice that if you still have an issue about something which is not the ``pip install`` you initially had, open a new issue.