guysoft/OctoPi #656: Improve build script output - location of images

supersebbo, 2020-06-19:
Could I suggest that the build scripts are improved to better locate the output.  Right now the resulting octopi image output is located in the src/workspace directory and is named the same as the input image.  This is counter-intuitive.

Similar build systems (such as OpenWRT) have a convention of placing the resulting images in a bin/ directory at the project root.  Putting build output in the source (src) directory is generally considered bad practice. 

Further it would be helpful for identification if the output image could have "-octopi" appended. 

i.e. if the input image is src/image/2020-05-27-raspios-buster-lite-armhf.zip
The output could be bin/2020-05-27-raspios-buster-lite-armhf-octopi.img

Thanks!

guysoft, 2020-06-21:
This is already done in the [release](https://github.com/guysoft/CustomPiOS/blob/devel/src/release) script.
If you are running using the Docker build method, the ``build`` command also runs the release script and will do exactly what you are describing.

guysoft, 2020-06-28:
Closing if no reply

closed 2020-08-09 by guysoft