guysoft/OctoPi #657: additional_brokenfps_usbdevices=(046d:0825)

midow, 2020-06-26:
#### What were you doing?

Logitech C270 USB camera would not work, fixed by activating the broken devices line.

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPi

0.17.0

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

guysoft, 2020-06-28:
Can you provide what you have added in the line for the camera to work?
Or even better - send a PR to the file: https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/filesystem/home/root/bin/webcamd#L18

midow, 2020-06-28:
I added the line shown in the subject. The Logitech C270 is 046d:0825

via phone

closed 2020-06-28 by guysoft

cxcorpx, 2022-01-07:
hi, i'm new in octoprint, i have 2 camera and both run slow fps, maybe u can help me to fix this i have the c920 and c930e i have 10 days reading post and nothing that can help me to fix this, if u can help i will be really happy i don't want to sell my cameras.

guysoft, 2022-01-08:
@cxcorpx I warmly suggest you ask in the OctoPrint community forum at: https://community.octoprint.org/

commit 71f9e549 (2020-06-28): Fixes #657