guysoft/OctoPi #666: Camera image red after upgrading OctoPi image from Jessie to Buster

wookie73, 2020-07-09:
Pi 3
Pi V2 camera

I upgraded from Jessie to Buster version of OctoPi Now my camera image is red.

Have tried changing the -awb to -awb tungsten and fluorescent:and off in octopi.txt. None of which seem to make any difference.
Also just tried changing the awb colour gain, --awbgainB & --awbgainY but they didn't seem to do anything either.

Have tried a different ribbon cable.

Seems to be a problem with the AWB.

Any help / ideas much appreciated.
Dave

[image]

tedder, 2020-07-09:
Assuming you don't live in a world where that's the normal color, [please post in the community forums](https://community.octoprint.org/latest). They're really good at helping debug things like this.

guysoft, 2020-07-09:
What version of OctoPi?

wookie73, 2020-07-09:
Have already posted in the community forums.
https://community.octoprint.org/t/camera-image-red-after-upgrading-octoprint-image-from-jessie-to-buster-pi-v2-cam/21891
Have also searched the  forum for other posts with the same issue. Two other posts but no answer. 
Have looked around a fair bit for an answer but cannot find one.

Seems that the -awb switches do not work in the octopi.txt file. That and the fact that everything worked fine with Jessie would indicate to me an issue with the Buster version on OctoPi.

Sorry missed version in,of out or original post.
OctoPrint version : 1.4.0
OctoPi version : 0.17.0

Thanks for the replies Dave

jantman, 2020-10-02:
I just happened by this, as I'm having the same issue. Do you happen to have a NoIR camera? Because this looks *exactly* like what a NoIR camera does with the wrong white balance. Here's an example of mine:

[image]

You should be able to confirm this by doing the following:

1. On the Pi, stop webcamd with ``sudo systemctl stop webcamd``
2. Take a snapshot of the camera with ``raspistill -o test.png`` - that should show the red/pink cast (view the image on the Pi if you can, or else scp it somewhere that you can see it)
3. Take a snapshot again with ``raspistill -awb greyworld -o greyworld.png`` and look at it. If I'm right, that should look normal.

Mine using "greyworld" settings":
[image]

If the second snapshot looks right, it's because your camera - like mine - needs the ``greyworld`` white balance mode. Unfortunately, the version of mjpg-streamer (webcamd) that's currently bundled with OctoPi doesn't support this mode. There's been a pull request ( https://github.com/jacksonliam/mjpg-streamer/pull/235 ) open for almost a year to fix that, but it hasn't been merged.

I've cut a [branch in my OctoPi fork](https://github.com/jantman/OctoPi/tree/rpi-cam-awb-greyworld) that pulls in the mjpg-streamer fork with these PR changes that enable the "greyworld" white balance mode, and enabled builds of it on GitHub. If you want to give it a try, a tarball of the ``/home/pi/mjpg-streamer`` directory with the patch applied, along with installation instructions, are at https://github.com/jantman/OctoPi/releases/tag/greyworld-v1

wookie73, 2021-02-02:
Amazing thanks.
Took 4 months for me to get around to looking at it but you solution did the trick.

Once again many thanks @jantman

closed 2021-02-02 by wookie73