guysoft/OctoPi #667: Free/Used Space is Blank on rc3. and Unable to successfully upload gcode.

dcodom, 2020-07-21:
#### What were you doing?

I am using OctoPrint rc3. I have ran into two issues....
I just upgraded from the latest stable release.

Since upgrading I no longer see what the drive space used or available is.
and
I just tried the drag and drop method along with the manual upload. And the screen never says that it is finished saving. If I refresh the browser, it shows the file in the list, but I am not sure it is all there.

#### Version of OctoPi
1.4.1rc3

#### Printer model & used firmware incl. version
Ender3 Pro 
RaspPi Version 3b+

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

dcodom, 2020-07-21:
Also when I hit the trashcan to delete the file I tried to upload, the trashcan just spins and spins.

dcodom, 2020-07-21:
fyi.... I just rolled it all back to 1.4.0 and it is all working as it should. I can go back to rc3 when/if this can be resolved.

guysoft, 2020-07-21:
This is an OctoPrint issue, not an OctoPi issue. Please report it to OctoPrint issue tracker.
https://github.com/foosel/OctoPrint/issues/

closed 2020-07-21 by guysoft