guysoft/OctoPi #669: Automatic update to Octoprint 1.4.0 with pip fails on OctoPi 0.17.0

coljac, 2020-08-03:
### What were you doing?

After a new install of OctoPi on a Pi 4B, OctoPrint advises that an update to 1.4.0 is available. Using the automatic updated and/or updating from the command line fails.

Steps to reproduce:

1. Insert fresh image of OctoPi into Pi 4B and edit the wifi settings.
2. Power up and SSH in, change password.
3. Navigate to octopi.local on another computer.
4. See "Update available" in top right, click on update. (OR: run `/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/foosel/OctoPrint/archive/1.4.0.zip --no-cache-dir` from the command line)

#### What did you expect to happen?

OctoPrint should have updated and restarted.

#### What happened instead?

Installation fails (in particular, the psutil dependency). The key part of `plugin_softwareupdate_console.log` is as follows (full log attached):

```2020-08-02 14:27:48,963 > Building wheels for collected packages: OctoPrint, psutil, future
2020-08-02 14:27:48,963 > Building wheel for OctoPrint (setup.py): started
2020-08-02 14:27:52,962 > Building wheel for OctoPrint (setup.py): finished with status 'done'
2020-08-02 14:27:52,963 > Created wheel for OctoPrint: filename=OctoPrint-1.4.0-py2.py3-none-any.whl size=3114116 sha256=b09338d1d99e07dd59a04ee3d25aae78ce5397b5a7cd23ef4526c1ee1c41e6d4
2020-08-02 14:27:52,963 > Stored in directory: /tmp/pip-ephem-wheel-cache-D4Rg0l/wheels/7a/eb/1a/846310e85c3df1e8b7fcdd9fa18de0f320a0335a7eb5f48451
2020-08-02 14:27:52,963 > Building wheel for psutil (setup.py): started
2020-08-02 14:27:55,429 > Building wheel for psutil (setup.py): finished with status 'error'
2020-08-02 14:27:55,430 > Running setup.py clean for psutil
2020-08-02 14:27:55,934 ! ERROR: Command errored out with exit status 1:
2020-08-02 14:27:55,934 ! command: /home/pi/oprint/bin/python2 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-_ePd4r/psutil/setup.py'"'"'; __file__='"'"'/tmp/pip-install-_ePd4r/psutil/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-4h5REg --python-tag cp27
2020-08-02 14:27:55,935 ! cwd: /tmp/pip-install-_ePd4r/psutil/
2020-08-02 14:27:55,936 ! Complete output (73 lines):
2020-08-02 14:27:55,936 ! during IPA pass: visibility
2020-08-02 14:27:55,936 ! /tmp/tmpVrE08h.c:1: internal compiler error: Segmentation fault
2020-08-02 14:27:55,936 ! #include <linux/ethtool.h>
2020-08-02 14:27:55,936 !
2020-08-02 14:27:55,937 ! 0xb6b1511f ???
2020-08-02 14:27:55,937 ! ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
2020-08-02 14:27:55,937 ! Please submit a full bug report,
2020-08-02 14:27:55,937 ! with preprocessed source if appropriate.
2020-08-02 14:27:55,937 ! Please include the complete backtrace with any bug report.
2020-08-02 14:27:55,938 ! See <file:///usr/share/doc/gcc-8/README.Bugs> for instructions.
2020-08-02 14:27:55,938 !
2020-08-02 14:27:55,938 ! /tmp/tmpVrE08h.c:1: internal compiler error: Illegal instruction
2020-08-02 14:27:55,938 ! 0xb6b1511f ???
2020-08-02 14:27:55,938 ! ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
2020-08-02 14:27:55,938 ! Please submit a full bug report,
2020-08-02 14:27:55,939 ! with preprocessed source if appropriate.
2020-08-02 14:27:55,939 ! Please include the complete backtrace with any bug report.
2020-08-02 14:27:55,939 ! See <file:///usr/share/doc/gcc-8/README.Bugs> for instructions.
2020-08-02 14:27:55,939 ! running bdist_wheel
2020-08-02 14:27:55,939 ! running build
```

#### Version of OctoPi

OctoPi 0.17.0
OctoPrint 1.3.12

#### Printer model & used firmware incl. version

Ender 3 Pro (but not relevant)

#### Screenshot(s)/video(s) showing the problem:

[plugin_softwareupdate_console.log](https://github.com/guysoft/OctoPi/files/5013755/plugin_softwareupdate_console.log)

guysoft, 2020-08-03:
@foosel seen anything like this on your upgrade rigs?
Is this a pi4B with how much ram?

foosel, 2020-08-03:
Have seen no such issues at all (and just last week I've updated several 0.17.0 installs again to various 1.4.x versions).

coljac, 2020-08-03:
> Is this a pi4B with how much ram?

4 gigs.

I will attempt some more installs from scratch and see if the problem is reproducible.

dertbv, 2020-08-05:
I had an upgrade failure as well.  Web interface would not come up.  Should I open another issue or paste my log here?

coljac, 2020-08-05:
I tried this once more with a fresh image of OctoPi and now upgrading to 1.4.1. It was successful. So either the original issue was a true quirk, or the dependencies for 1.4.1 are slightly different allowing pip to complete. This can be closed.

tedder, 2020-08-06:
FWIW the 4B4gb has some specific issues not seen in other 4Bs.  But I suspect this is related to undervoltage/overheating. dertbv and any others, please post full diagnostic info- the upgrade log as well as the output from `vcgencmd get_throttled`, the output from `tail -8 /proc/cpuinfo`, the octoprint.log, etc.

dertbv, 2020-08-06:
[octoprint.log](https://github.com/guysoft/OctoPi/files/5038478/octoprint.log)
I have already reverted back to 1.4.0  this is the only log that i kept.

guysoft, 2020-10-02:
Can I close this?
Closing if no reply