guysoft/OctoPi #670: [Request]

cp2004, 2020-08-07:
Request: OctoPi nightly builds have some sort of dev or nightly tag on them.

Reasoning behind this: 
I'm writing a script to upgrade your OctoPrint Python environment to Python 3, ([Which you can see here](https://github.com/cp2004/Octoprint-Upgrade-To-Py-3)) for existing users. 
I have implemented a check of OctoPi version numbers, and was only allowing versions 0.16 and 0.17 to upgrade - earlier versions missed some dependencies, 0.18 brings Python 3 as standard. But, as pointed out to me, not all 0.18 versions have Python 3 as standard, so I have had to allow them to update, but with a Python version check.

Basically, what I'm saying is there is no way of telling (to my knowledge, correct me if there is) at what point the nightly build was made, and as a result it is harder for me to tell what I need to do.

Something like `0.18.0-nightly.2020.08.06`, so you can tell the difference. Or you can go shorter and go with `0.18.0-nightly.1` and increment manually, but I thought the date might be able to be computed by a github action of sort.

cp2004, 2020-08-07:
I gave this a shot on my fork, just overwriting the config file once checked out in the runner - haven't had a chance to flash it and confirm the version is changed, and then squash and PR. If you want to see, I did it here https://github.com/cp2004/OctoPi

The image built successfully, but not tested. And it would involve changing the action to increment version number, which is not ideal.

guysoft, 2020-08-09:
Hey, the zip file holds the date the nightly was build. eg: `https://storage.googleapis.com/octoprint/2020-08-06_2020-05-27-octopi-buster-lite-armhf-0.18.0.zip``
Note the start date: ``2020-08-06`` then ``_``

I am not changing the img name because that needs to remain ``0.18.0`` for the release.

cp2004, 2020-08-09:
I see what you mean, and noticed that the github actions don't actually upload anywhere, assume its just for testing now? 
However, and I see this was not explained well orginally, I mean the version number in `/etc/octopi_version`, is there a way to tell at what point the image was built?

Ideally, the only case for this is to tell at what point features were added. In my script, (linked in original issue) I wanted to query version number, to tell if it was Python 3 by default. So I basically put `version < 0.18`, however not all `0.18` builds have Python 3 installs. Just an idea, as otherwise there are `0.18` builds that have different features, so it is not a good indicator

guysoft, 2020-08-09:
The github action is new and we didn't even add documentation for it. The nightly in the readme points to the current nightly builds. As I said, we don't use it for nightly builds exactly because its not fully suited for how OctoPi works yet. If you want to sort up a PR that builds nightly to it then it might be relevant. Also, we didn't test download speeds and retention comparing to the current nightly builds.
See #661

guysoft, 2020-10-02:
Closing if no reply

cp2004, 2020-10-02:
Well there's no resolution to this, since the original issue has not been addressed, so I was expecting a reply.

There's no way to tell the difference between a nightly build from December (way before they are Python 3) and one built yesterday - but the feature set is far different. And I don't mean when they are downloaded, I mean once they are installed. 

As a developer, it is no use to me what the `.img` file is named, when someone downloads it, only the internal `0.18`

My request was, to add 'nightly' or something to the version tag, so I can tell the difference from the 'stable release'.

If you don't want to do this, by all means close it just let me know it will not be worked on.