guysoft/OctoPi #673: Endstop switches are not being registered properly.

bholzer, 2020-09-01:
#### What were you doing?

Homing the printer.

#### What did you expect to happen?

I expected the carriage to stop when it contacted the endstop.

#### What happened instead?

The endstop switches are not registering.

I am sending `M119` while holding down my endstop switches, and only the y axis switch seems to be reporting as "triggered" when pressed. So, it seems that the endstop switches aren't being registered at all when OctoPrint/OctoPi is connected. The homing works fine without OctoPi connected, so I can be certain the switches are not broken and that the firmware can read them.

After I home the printer and then connect OctoPi:

```
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: TRIGGERED
Recv: z_min: open
```

I am not sure how to troubleshoot further.

#### Version of OctoPi

OctoPi: 0.17
OctoPrint: 1.4.2

#### Printer model & used firmware incl. version

Ender 3 v2, Marlin 2.0.6.1

#### Browser and version of browser, operating system running browser

Chrome  84.0.4147.135, windows 10

#### Link to octoprint.log

https://gist.github.com/bholzer/fec187ab47cd70317c775b6990d9b1a9

I have read the FAQ.

guysoft, 2020-09-01:
This is an OctoPrint issue, not an OctoPi issue, please post it to the OctoPrint community at: https://discourse.octoprint.org/
Also - that recv message is coming from your firmware, so I would bet its a firemware issue.

bholzer, 2020-09-01:
I thought the same thing, but the same issue was filed with OctoPrint, and they claim there is simply no way for OctoPrint to be causing this issue. 

Link here: https://github.com/OctoPrint/OctoPrint/issues/3714

A firmware problem makes sense in this context I suppose, but I am really confused about how it would work in some cases, and then suddenly stop working when connected to the Pi. I used the example configuration provided by Marlin: https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3%20V2

I am new to printer firmware, so I don't really know where to begin looking.

guysoft, 2020-09-28:
I am not really sure how to help you without the hardware.
It might have stopped working because something got fried, I had that several time. You might need to validate that with a multimeter

guysoft, 2020-09-29:
I see this was solved on marlin, should I close?