guysoft/OctoPi #674: Raspberry Pi Camera Module v2 is considered as USB webcam

yschroeder, 2020-09-03:
#### What were you doing?

1. Connect a Raspberry Pi Camera Module v2 to the Pi running OctoPi
2. Change the `camera_raspi_options` in `/boot/octopi.txt`
3. Changes have no effect after restarting `webcamd`

#### What did you expect to happen?

Changes to `camera_raspi_options` change the output of `mjpg_streamer` in the OctoPrint UI.

#### What happened instead?

Camera picture was not affected by the options.

#### Did the same happen when running OctoPrint in safe mode?

Did not test, as this is not a problem of OctoPrint.

#### Version of OctoPi

0.17.0

#### Relevant output of `journalctl -u webcamd`

```
Sep 03 06:53:41 octopii webcamd[26541]: Starting up webcamDaemon...
Sep 03 06:53:41 octopii webcamd[26541]: --- Configuration: ----------------------------
Sep 03 06:53:41 octopii webcamd[26541]: cfg_file:      /boot/octopi.txt
Sep 03 06:53:41 octopii webcamd[26541]: camera:        auto
Sep 03 06:53:41 octopii webcamd[26541]: usb options:   -r 640x480 -f 10
Sep 03 06:53:41 octopii webcamd[26541]: raspi options: -fps 10 -ISO 800 -mm backlit -br 100 -hf
Sep 03 06:53:41 octopii webcamd[26541]: http options:  -w ./www-octopi -n
Sep 03 06:53:41 octopii webcamd[26541]: Explicitly USB device:
Sep 03 06:53:41 octopii webcamd[26541]: -----------------------------------------------
Sep 03 06:53:41 octopii webcamd[26541]: Found video devices:
Sep 03 06:53:41 octopii webcamd[26541]: /dev/video0
Sep 03 06:53:41 octopii webcamd[26541]: /dev/video10
Sep 03 06:53:41 octopii webcamd[26541]: /dev/video11
Sep 03 06:53:41 octopii webcamd[26541]: /dev/video12
Sep 03 06:53:41 octopii webcamd[26541]: /dev/video13
Sep 03 06:53:41 octopii webcamd[26541]: /dev/video14
Sep 03 06:53:41 octopii webcamd[26541]: /dev/video15
Sep 03 06:53:41 octopii webcamd[26541]: /dev/video16
Sep 03 06:53:41 octopii webcamd[26541]: raspi
Sep 03 06:53:41 octopii webcamd[26541]: config file='/boot/octopi.txt':USB device was not set in options, start MJPG-streamer with the first found video device: /dev/video0
Sep 03 06:53:41 octopii root[26553]: Starting USB webcam
Sep 03 06:53:41 octopii webcamd[26541]: <13>Sep  3 06:53:41 root: Starting USB webcam
Sep 03 06:53:41 octopii webcamd[26541]: Running ./mjpg_streamer -o output_http.so -w ./www-octopi -n -i input_uvc.so -r 640x480 -f 10 -d /dev/video0
Sep 03 06:53:41 octopii mjpg_streamer[26554]: MJPG-streamer [26554]: starting application
Sep 03 06:53:41 octopii mjpg_streamer[26554]: MJPG-streamer [26554]: MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
Sep 03 06:53:41 octopii webcamd[26541]: MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
Sep 03 06:53:41 octopii webcamd[26541]:  i: Using V4L2 device.: /dev/video0
Sep 03 06:53:41 octopii webcamd[26541]:  i: Desired Resolution: 640 x 480
Sep 03 06:53:41 octopii webcamd[26541]:  i: Frames Per Second.: 10
Sep 03 06:53:41 octopii webcamd[26541]:  i: Format............: JPEG
Sep 03 06:53:41 octopii webcamd[26541]:  i: TV-Norm...........: DEFAULT
Sep 03 06:53:41 octopii mjpg_streamer[26554]: MJPG-streamer [26554]: Using V4L2 device.: /dev/video0
Sep 03 06:53:41 octopii mjpg_streamer[26554]: MJPG-streamer [26554]: Desired Resolution: 640 x 480
Sep 03 06:53:41 octopii mjpg_streamer[26554]: MJPG-streamer [26554]: Frames Per Second.: 10
Sep 03 06:53:41 octopii mjpg_streamer[26554]: MJPG-streamer [26554]: Format............: JPEG
Sep 03 06:53:41 octopii mjpg_streamer[26554]: MJPG-streamer [26554]: TV-Norm...........: DEFAULT
```

#### Workaround

Set `camera="raspi"` instead of `camera="auto"` in `/boot/octopi.txt` results in the changes to `camera_raspi_options` to be applied and the log of `webcamd` changes to the following:

```
Sep 03 06:54:55 octopii webcamd[26717]: Starting up webcamDaemon...
Sep 03 06:54:55 octopii webcamd[26717]: --- Configuration: ----------------------------
Sep 03 06:54:55 octopii webcamd[26717]: cfg_file:      /boot/octopi.txt
Sep 03 06:54:55 octopii webcamd[26717]: camera:        raspi
Sep 03 06:54:55 octopii webcamd[26717]: usb options:   -r 640x480 -f 10
Sep 03 06:54:55 octopii webcamd[26717]: raspi options: -fps 10 -ISO 800 -mm backlit -br 100 -hf
Sep 03 06:54:55 octopii webcamd[26717]: http options:  -w ./www-octopi -n
Sep 03 06:54:55 octopii webcamd[26717]: Explicitly USB device:
Sep 03 06:54:55 octopii webcamd[26717]: -----------------------------------------------
Sep 03 06:54:55 octopii webcamd[26717]: Found video devices:
Sep 03 06:54:55 octopii webcamd[26717]: /dev/video0
Sep 03 06:54:55 octopii webcamd[26717]: /dev/video10
Sep 03 06:54:55 octopii webcamd[26717]: /dev/video11
Sep 03 06:54:55 octopii webcamd[26717]: /dev/video12
Sep 03 06:54:55 octopii webcamd[26717]: /dev/video13
Sep 03 06:54:55 octopii webcamd[26717]: /dev/video14
Sep 03 06:54:55 octopii webcamd[26717]: /dev/video15
Sep 03 06:54:55 octopii webcamd[26717]: /dev/video16
Sep 03 06:54:55 octopii webcamd[26717]: raspi
Sep 03 06:54:55 octopii webcamd[26717]: config file='/boot/octopi.txt':Start MJPG-streamer with video device: raspi
Sep 03 06:54:55 octopii root[26728]: Starting Raspberry Pi camera
Sep 03 06:54:55 octopii webcamd[26717]: <13>Sep  3 06:54:55 root: Starting Raspberry Pi camera
Sep 03 06:54:55 octopii webcamd[26717]: Running ./mjpg_streamer -o output_http.so -w ./www-octopi -n -i input_raspicam.so -fps 10 -ISO 800 -mm backlit -br 100 -hf
Sep 03 06:54:55 octopii mjpg_streamer[26729]: MJPG-streamer [26729]: starting application
Sep 03 06:54:55 octopii mjpg_streamer[26729]: MJPG-streamer [26729]: MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
Sep 03 06:54:55 octopii webcamd[26717]: MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
Sep 03 06:54:56 octopii webcamd[26717]:  i: fps.............: 10
Sep 03 06:54:56 octopii webcamd[26717]:  i: resolution........: 640 x 480
Sep 03 06:54:56 octopii webcamd[26717]:  i: camera parameters..............:
Sep 03 06:54:56 octopii webcamd[26717]: Sharpness 0, Contrast 0, Brightness 100
Sep 03 06:54:56 octopii webcamd[26717]: Saturation 0, ISO 800, Video Stabilisation No, Exposure compensation 0
Sep 03 06:54:56 octopii webcamd[26717]: Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Sep 03 06:54:56 octopii webcamd[26717]: Metering Mode 'backlit', Colour Effect Enabled No with U = 128, V = 128
Sep 03 06:54:56 octopii webcamd[26717]: Rotation 0, hflip Yes, vflip No
Sep 03 06:54:56 octopii webcamd[26717]: ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
Sep 03 06:54:56 octopii webcamd[26717]:  o: www-folder-path......: ./www-octopi/
Sep 03 06:54:56 octopii webcamd[26717]:  o: HTTP TCP port........: 8080
Sep 03 06:54:56 octopii webcamd[26717]:  o: HTTP Listen Address..: (null)
Sep 03 06:54:56 octopii webcamd[26717]:  o: username:password....: disabled
Sep 03 06:54:56 octopii webcamd[26717]:  o: commands.............: disabled
Sep 03 06:54:56 octopii mjpg_streamer[26729]: MJPG-streamer [26729]: fps.............: 10
Sep 03 06:54:56 octopii webcamd[26717]:  i: Starting Camera
```

Note: There was not USB webcam connected at any time to the Pi, it always uses the Raspberry Pi Camera Module v2!

gotdsm95, 2020-09-15:
Are you using The Spaghetti Detective?

guysoft, 2020-09-15:
@gotdsm95 Does Spaghetti Detective change anything?

yschroeder, 2020-09-15:
@gotdsm95 I am not, and I also can't imagine how that would change anything.

gotdsm95, 2020-09-15:
It uses another type of streaming that can't coinside with mjpeg streamer. 
https://www.thespaghettidetective.com/docs/webcam-feed-is-laggy/

marcolivierroy, 2020-09-17:
The camera gets recognized as a USB camera using the bcm2835-v4l2 driver. It is a normal behavior.

yschroeder, 2021-01-29:
I guess #690 addresed this and this can be closed.

closed 2021-01-31 by guysoft