guysoft/OctoPi #677: [Request] Enhance webcamd & /boot/octopi.txt to allow for multiple cameras

kazooless, 2020-09-24:
Hello,

I have figured out how to get mjpg_streamer to run 4 instances with just one command line. I am too much of a beginner when it comes to Bash scripting though, so I personally am not able to edit the webcamd script to handle the logic to see if there are more than 1 video devices to use. So as a feature request I am suggesting enhancing the scripts to handle more than one.

Chris Riley has a video series explaining how to setup multiple instances of Octoprint using Octopi but he says in it no matter what he tries, he can't get 4 webcam instances. This would be an alternate way (and I suspect it would use up fewer resources too). The important part though is using UDEV rules to permanently call the same camera for its assigned printer. So generic logic looking to see if multiple cameras exist isn't enough.

Here is the command line that works for me: 

`sudo ./mjpg_streamer -o "output_http.so -l octopi4.local -p 8083 -w ./www" -i "input_raspicam.so -fps 10 -x 1920 -y 1080" -i "input_uvc.so -d /dev/wcam-1080 -fps 10 -r 1920x1080" -i "input_uvc.so -d /dev/wcam-Logitech -fps 10 -r 1920x1080" -i "input_uvc.so -d /dev/wcam-ARC -fps 10 -r 1920x1080”`

I am successfully using 1 raspberry pi camera & 3 USB cameras with this one command line.

I would suggest changing /boot/octopi.txt to have 4 sections with only the first one uncommented. Sections 2-4 could then be uncommented as needed, and filled out with the new device path for each camera. Then in the webcamd script, I would imagine a for loop of some sort to check for how many cameras are uncommented and then going from there to form the command line in the startcamera function.

guysoft, 2020-09-25:
A PR would be welcome,  too busy recently but would welcome this if its added in.

kazooless, 2020-09-25:
@guysoft 

I completely understand. I am definitely willing to put in the effort, but I'm not so sure I'm able. Is there any chance you could give me a rough skeleton of comments outlining the logic needed? I think I could figure out and write the code based off of something like that. If not, I'll try anyway, but I am definitely just a beginner with only self-taught knowledge on scripting.

Here is what I have so far for a new /boot/octopi.txt:

`#!/bin/bash

### Webcam Options

#PICAM=TRUE     ##  For a Raspberry Pi camera, uncomment

##  Assigned webcams. For each USB webcam, uncomment & edit the custom device name.
##  It is not necessary to start with WEBCAM1 (e.g. if you have a raspi camera you want to be 1st)

#WEBCAM1=/dev/wcam-1
#WEBCAM2=/dev/wcam-2
#WEBCAM3=/dev/wcam-3
#WEBCAM4=/dev/wcam-4

### Additional options to supply to MJPG Streamer for the USB camera
#
# See https://faq.octoprint.org/mjpg-streamer-config for available options
#

# Webcam resolution
# Defaults to a resolution of 640x480 px

#RESPI="-x 1920 -y 1080"        ## Raspberry Pi resolution
#RES1="1920x1080"               ## Webcam 1 resolution
#RES2="1920x1080"               ## Webcam 2 resolution
#RES3="1920x1080"               ## Webcam 3 resolution
#RES4="1920x1080"               ## Webcam 4 resolution

# Frame Rate
# Defaults to a framerate of 10 fps

#FRPI="10"                       ## Raspberry Pi Frame Rate
#FR1="10"                        ## Webcam 1 Frame Rate
#FR2="10"                        ## Webcam 2 Frame Rate
#FR3="10"                        ## Webcam 3 Frame Rate
#FR4="10"                        ## Webcam 4 Frame Rate

`

timur-tabi, 2021-03-14:
Instead of having four sections, a better idea would be to have multiple files, like how /etc/init.d works.

I do think that support for multiple USB cameras is long overdue in OctoPi.