guysoft/OctoPi #680: [Request] Mjpg streamer -raspistill request for octopi/print

carcamerarig, 2020-09-30:
Hi Guy, octo team. 
I have a feature request (I think anyway) but first I have to say excellent work by you and your team with the fact everything just works. 

So I had been striving for HQ timelapse video but became vexed at the still image quality in the stream, as I understand it the still image assets are stripped from the video frames at X video resolution, due to bandwidth I have found anything above 720p struggles to stream. I’m using pi4b8gb and a Pi cam v2.1 with octolapse, so following Liam Jackson’s  mjpg-streamer thread on the pi foundation forums, raspistill was a thing a while back but forked?? and ended up using the video port instead.
So my question is, is it possible (if so what would need to happen?) to have mjpg-streamer use raspistill on the camera port at  3280 × 2464 pixels instead of video at all?
I don’t understand  the need for video at all as the service is for generating timelapse made up of still images which get rendered into a video.
Could it be possible to code and use the raspistill service instead of video, at the full res 3280 × 2464 pixels  which could easily be downscaled during render and the frame rate of 1 fps used to generate the preview in octo. 1 fps would be ample for a per layer trigger wouldn’t it?
Apologies if this request isn’t in right place, I’m not really sure where best to put it for the right people to see it. 

Thanks
Justin.

carcamerarig, 2020-10-01:
Anyone?

guysoft, 2020-10-02:
1.You can ask that on [mjpg-streamer](https://github.com/jacksonliam/mjpg-streamer)
2. Downscaling is a heavy process, you won't be able to have that running in realtime.
3. Take a look here, people are saying they did manage to do this: https://community.octoprint.org/t/octolapse-resolution-with-raspi-cam/3368/13

carcamerarig, 2020-10-02:
> 1.You can ask that on [mjpg-streamer](https://github.com/jacksonliam/mjpg-streamer)
>
> 2. Downscaling is a heavy process, you won't be able to have that running in realtime.
>
> 3. Take a look here, [...]

Thanks for the reply, I have been asking on mjpg streamer also been looking at the forks and timeline. The downscaling I just a thought l, ultimately I just want hi res images I can process in a dedicated software later if need be. Jacksonliam appears to not have been logged for a considerable amount time, I wondered if he'd abandoned his project?? Looking back through the timeline mjpg streamer was using the raspistill port, would it be difficult for someone to code a fork that just uses raspistill on the camera port at 1 fps,even if all it does is stored the hi res images in a folder for download?

carcamerarig, 2020-10-05:
Could there be some work around to have the raspi cam function exactly like tge external camera function? It grabs a still image at the cameras native setup and drops the image into the snapshot folder. That's all I'm after.

guysoft, 2020-10-05:
> Could there be some work around to have the raspi cam function exactly like tge external camera function? 
Not that I know of

carcamerarig, 2020-10-05:
Right ok, what part of octoprint would deal that? So I can at least start to explore those avenues. 

Many Thanks

Justin Edgington
Visit us: Car camera Rig
Email us: justin@carcamerarig.com

Call us: +44 (0) 7973 746192

guysoft, 2020-10-28:
Just saw now.
The strea runs on mjpg streamer. The service is webcamd:
https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/filesystem/root/etc/systemd/system/webcamd.service

There is an option to select a different streamer at:
https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/filesystem/root/etc/systemd/system/streamer_select.service

The script that runs that is at:
https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/filesystem/home/root/bin/streamer_select

Hope that helps