guysoft/OctoPi #682: Unknown IP Addresses

Doubledeucedavis, 2020-10-11:
My wifi automatically shut down my RaspberryPi because it was detecting unknown IP addresses. Anywhere from Russia to Poland to Canada. I reset my Pi and I do not want to reinstall Octoprint because of the threats my Wifi blocked. It may have been a plugin but I do not have a list of the plugins I had.

Doubledeucedavis, 2020-10-11:
Any help is greatly appreciated. I love using it and it streamlines everything.

guysoft, 2020-10-11:
I am not sure how OctoPi could be related to this.
You could install a firewall on the RaspberryPi and configure it to receice only local network traffic:
Also, you might have a plugin that is making remote requests.
https://www.raspberrypi.org/documentation/configuration/security.md

@foosel saw anything like that?

foosel, 2020-10-11:
> @foosel saw anything like that?

Nope. Sounds unrelated

Doubledeucedavis, 2020-10-11:
Sorry for taking away from your time. I decided to re install octopi and not install many plugins.

closed 2020-10-11 by Doubledeucedavis