guysoft/OctoPi #683: open appology to foosel with no request or expectation for forgiveness

belenaetcheriscancer, 2020-10-16:
@foosel 
 i had no idea you were even one of the project programmers my fault. i spent days learning linux and trying to install raspian/octoprint 
, i pressumed because of my own lack of knowledge with github that you were just a forum user and you were project lead for the tutorial that i wrongfully thought the team were uninterested in filling out to the full crossing all t's and dotting all the i's , there was no right reason for my anger , covids obviously lead me to be volatile and unfiltered when irritated and i directed my anger out at you a stranger for your tutorial not being up to my own standards which shouldnt have been imposed on anyone accept for myself when teaching someone else , because of my own anger towards past bad teachers i let a tutorial get to me which was low of me , its not your first language and i should have taken that into account but i expelled my frustration at you instead of using that frustration for something beneficial i probably shat on your day which you didnt deserve , Ive done my research and you are a good person proven by your posts and excursions from octoprint to help other projects and im sorry if i ruined you day or even you're hour i should'nt have let a few wasted days get in the way of compasion towards others or being constructive instead of destructive personally if you rip me down ill build myself back up stronger im lucky in that sense and i forget others are not so lucky 
so to you and the community i appologize and have no expectation of coming back to any open arms as my conduct does not deserve it

Jnesselr, 2020-10-16:
Hi there! I'm just gonna insert myself into this conversation real quick and point out some problematic things here:

1. This isn't even OctoPrint's repo. It's OctoPi which is the repo used to make Raspberry Pi images.
2. "one of" my dude, she is THE programmer. But even if she wasn't, that's no way to treat people. At all. COVID is no excuse, we're all going through it.
3. Get some therapy. If you've got past issues with bad teachers, you need to talk to someone qualified and resolve them, not blow up at someone on the internet. I'm serious, see a therapist, it'll help.
4. This doesn't come across as an apology. It doesn't matter if it's her first language or not. This is a post filled with self flagellation and it's all about you. I mean the "if you rip me down" comment like... what? She's not lucky in that she can't take a steaming pile of abuse and feel better because of it? Again. Ya need therapy. (that's not a bad thing, everyone needs therapy, but this is an issue that's obviously causing problems for you and others)
5. Your username is shitting on another project.
6. When people like you hurt my friend, I hear about it. I don't like my friend hurting which is why you're hearing from me.

If you're going to apologize, do it coherently, explain the harm you did, take responsibility for it, don't make excuses, don't self flagellate, explain how you're going to fix things (have I suggested therapy?), and follow through with that. There's some good tips on authentic apologies that I suggest you check out like https://blogs.psychcentral.com/love-matters/2018/06/the-top-12-fake-apologies-and-what-makes-for-an-authentic-apology/

foosel, 2020-10-16:
Your apology here basically boils down to "I had no idea who you were, so I figured it wasn't that bad to vent my frustration as a full on newbie about struggling with something in the most insulting way I could think of, and also that was just due to COVID, also I'm sorry your grasp of the english language is the way it is, you are not a native speaker".

Even your nickname here is an insult to the hard work of yet another good project. 

You should spend some time with yourself, learn to control your anger, learn how to communicate with people. Maybe that will help you grow. Personally though I have absolutely no interest in further interaction with you (frankly your way of writing is also not up to MY own standards and gives me a headache), and this is the wrong place for all of this anyhow.

As I told you before banning you for your absolutely intolerable DMs on the forums last night: Go play somewhere else. You are no longer welcome in the OctoPrint community.

@guysoft I'm sorry this somehow ended up taking up space in your repository, sadly this was outside of my control.

closed 2020-10-16 by foosel

nallath, 2020-10-16:
>     5. Your username is shitting on another project.

It's not just the username that was doing it. Just check the issue that was opened there +- 3 hours ago. It has some very colorfull language that I don't care to repeat here.