guysoft/OctoPi #689: Communication timeout while printing,

t3uch1, 2020-10-27:
#### What were you doing? Print with octolapse

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead? Print stop with error : 
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.

#### Did the same happen when running OctoPrint in safe mode?
Octolapse is not functional in safe mode

#### Version of OctoPi
1.4.2

#### Printer model & used firmware incl. version
Two trees Bluer plus

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

tedder, 2020-10-27:
This isn't reproducible. It doesn't provide any information (such as a serial log). For all we know you cut your USB cable midway through the print.

You said "print with octolapse", does it happen when you disable octolapse? Print in safe mode?

t3uch1, 2020-10-27:
I did several tests, with and without octolapse, the error this product has different places from the print

I tried 4 times and 4 times it happened

I tried several USB cables, to cut the Gcode differently etc ...

This cuts after about 4 or 6 hours of print

cp2004, 2020-10-28:
This is the repository for OctoPi, the operating system image with OctoPrint preinstalled. You need to find support in the community forum or with the OctoLapse author.

guysoft, 2020-10-28:
What @cp2004 said. 
Links 
https://discourse.octoprint.org/
https://github.com/FormerLurker/Octolapse/issues

closed 2020-10-28 by guysoft