guysoft/OctoPi #692: OctoPi 0.18.0 RC1 Status

guysoft, 2020-11-04:
First release candidate for OctoPi 0.18.0

There are both 32bit and 64bit Based on RpiOS images available. Which give support to Raspberry Pi 4B 8GB and Raspberry Pi 400

The main change that would be likely noticable to plugin developers is that OctoPi comes with Python3 now.

Please try the release candidate.

32bit armf:
Download it at:
http://unofficialpi.org/Distros/OctoPi/2020-11-02_2020-08-20-octopi-buster-armhf-lite-0.18.0.zip

Md5: [``d3cb6af431e326cb8bd27719016d3922``](http://unofficialpi.org/Distros/OctoPi/2020-11-02_2020-08-20-octopi-buster-armhf-lite-0.18.0.zip.md5).

64bit amd64/aarch64:
Download it at:
http://unofficialpi.org/Distros/OctoPi/2020-11-02_2020-08-20-octopi-buster-arm64-lite-0.18.0.zip

Md5: [``643a6f4da6170607c57769a0c5bf2e7e``](http://unofficialpi.org/Distros/OctoPi/2020-11-02_2020-08-20-octopi-buster-arm64-lite-0.18.0.zip.md5).

### Changes in the image

* OctoPrint 1.4.2
* ``/home/pi/scripts/safemode`` (thanks @OutsourcedGuru )
*  Fix #619 Shutdown requires network (Thanks @da4id )
* #623 from szopen111/fix-issue-621 (Workaround for low resolution cameras)
* Fix/enable haproxy logging #640 and enable compression at haproxy layer #643 (thanks @tedder )
* Experimental HLS support #650 #668 #688 (thanks @chudsaviet )
* Removed old Cura.  See #654 and #444 .
* Some usb cameras setting fixed (#657 https://github.com/guysoft/OctoPi/commit/b64a5441b807f449417b8bcedb41b2703adf89e9 https://github.com/guysoft/OctoPi/commit/0d0ea3751e3866d7f893699894691025e33a7757 )
* OctoPi now uses Python3
*  Fixes #676 and adds deps to install numpy from wheel for plugins taht need numpy
* New arm64 builds
* Set HAProxy Diffie-Hellman key size to 2048 #685

### Build notes
* Uses [CustomPiOS 1.1.0](https://github.com/guysoft/CustomPiOS/releases/tag/1.4.0)

carl5765, 2020-11-05:
im running multiple instances on one pi will this upgrade include upgrading them as when i install python 3 it doesn't see my second or third image

Prutsium, 2020-11-05:
Will the 64bit also work in the 2GB Pi4?
Have to setup a second Pi anyway as i cant get multiple instances to work (all guides are based on the 0.17 version that seems to have /etc/default/octoprint but 0.18 doesnt have that :).

guysoft, 2020-11-05:
@Prutsium 
Should work with any device that supports 64bit instruction set. That includes all Pi4 and also 3B+

guysoft, 2020-11-05:
@carl5765 I didn't understand. This is an OctoPi image, not an OctoPrint package upgrade. You need to physically flash your SD card.

derekslenk, 2020-11-05:
Any way to update a current instance to this or just format/backup and restore

dxgldotorg, 2020-11-05:
@guysoft should the 64-bit image be used on a minimum 4GB system?

LMS0815, 2020-11-06:
> First release candidate for OctoPi 0.18.0
> 
> There are both 32bit and 64bit Based on RpiOS images available. Which give support to Raspberry Pi 4B 8GB and Raspberry Pi 400
> 
> The main change that would be likely noticable to plugin developers is that OctoPi comes with Python3 now.
> 
> Please try the release candidate.
> 
> 32bit armf:
> Download it at:
> http://unofficialpi.org/Distros/OctoPi/2020-11-02_2020-08-20-octopi-buster-armhf-lite-0.18.0.zip
> 
> Md5: [`d3cb6af431e326cb8bd27719016d3922`](http://unofficialpi.org/Distros/OctoPi/2020-11-02_2020-08-20-octopi-buster-armhf-lite-0.18.0.zip.md5).
> 
> 64bit amd64/aarch64:
> Download it at:
> http://unofficialpi.org/Distros/OctoPi/2020-11-02_2020-08-20-octopi-buster-arm64-lite-0.18.0.zip
> 
> Md5: [`643a6f4da6170607c57769a0c5bf2e7e`](http://unofficialpi.org/Distros/OctoPi/2020-11-02_2020-08-20-octopi-buster-arm64-lite-0.18.0.zip.md5).
> 
> ### Changes in the image
> * OctoPrint 1.4.2
> * OctoPi now uses Python3
> * `/home/pi/scripts/safemode` (thanks @OutsourcedGuru )
> * Fix #619 Shutdown requires network (Thanks @da4id )
> * #623 from szopen111/fix-issue-621 (Workaround for low resolution cameras)
> * Fix/enable haproxy logging #640 and enable compression at haproxy layer #643 (thanks @tedder )
> * Experimental HLS support #650 #668 #688 (thanks @chudsaviet )
> * Removed old Cura.  See #654 and #444 .
> * Some usb cameras setting fixed (#657 [b64a544](https://github.com/guysoft/OctoPi/commit/b64a5441b807f449417b8bcedb41b2703adf89e9) [0d0ea37](https://github.com/guysoft/OctoPi/commit/0d0ea3751e3866d7f893699894691025e33a7757) )
> * Fixes #676 and adds deps to install numpy from wheel for plugins taht need numpy
> * New arm64 builds
> * Set HAProxy Diffie-Hellman key size to 2048 #685
> 
> ### Build notes
> * Uses [CustomPiOS 1.4.0](https://github.com/guysoft/CustomPiOS/releases/tag/1.4.0)

Working like a charm on RASPBERRY PI 3B+ Raspberry Pi 3 B+, 4x 1,4 GHz, 1 GB RAM, WLAN, BT
Thank you!

guysoft, 2020-11-06:
@derekslenk Yes see below your comment.
@dxgldotorg No see above your comment.

dorfman2, 2020-11-06:
Had a very successful transition on a Raspberry pi 4 (1gig). Backup was seamless as well.

PistolsAtDawn, 2020-11-07:
Hi! Love the software! 

Just wanted to say that I cannot get my PS3 camera to work out of the box on 0.18 RC1. I didn't have time to test why this is though. It works flawlessly on 0.17.

Thanks!

sohxet, 2020-11-08:
Sending an M997 command to reboot the printer after uploading a firmware file through marlin-bft plugin causes octoprint to become unresponsive, pi doesn't respond to ssh connections. Raspi 4B 4GB, SKR V1.3 mainboard running marlin 2.0.6.1 bugfix.  
[octoprint.log](https://github.com/sohxet/octopi-0.18-testing/blob/master/octoprint.log)

Will-wastelander, 2020-11-09:
Back and restore from 0.17.0 to 0.18.0 RC1 went w/o a hitch on my rPi 4 4GB. I have 2 more to do. Currently running multiple plugins that were imported, and also installed OctoDash for screen use. Will be doing the same for the other 2 systems.

rbrian, 2020-11-09:
Not sure what happened... but... I have an external USB SSD attached to my Pi that held my uploads, timelapses, etc. Initial setup was easy several months ago, just formatted it via SSH and it mounted automatically. Has been working fine ever since. Today I decided it was time to bite the bullet and upgrade to Python 3, and figured I'd just use the RC to do it. So I backed up via the OctoPi settings page, and copied my home directory on the Pi to the SSD so that I'd have all my other files, like my Klipper config. I flashed the RC just fine, and everything seemed to be OK. (I am having a bit of trouble with it not finding octopi.local for SSH, but it works via the IP address, so I'm not too concerned.) Then I went to go look on the SSD for my SAMBA config file... and there's nothing. Drive won't automount, and is empty when I mount manually. File recovery tools aren't helpful. This just cost me a week or so of tinkering with the Klipper configuration to get it working right & calibrated again. :( If anyone has ANY clue what might have happened, I'll be grateful to hear about it. I'm going to leave the SSD offline for the time being, just in case there is actually some form of recovery that will help. 

I'll keep on with the RC, and let you know if I find anything else out of the ordinary.

sohxet, 2020-11-11:
OctoPrint 1.5.0rc1 was released, does this affect testing of octopi 0.18?

guysoft, 2020-11-11:
Talk to @foosel she mentioned it. Didn't seem to bother her.
Just making sure. You rather not wait?

foosel, 2020-11-11:
Considering that people were running into issues with current Pi4s vs the 0.17 image and the Pi400 also having gotten released (which I rightfully anticipated people would ask for support for, even though I don't understand WHY), I figured it would make more sense to push an OctoPi image RC now with the current stable OctoPrint on it, rather than wait until 1.5.0 hits stable (which depending on how this RC phase goes might still take until December). I stand by this decision.

guysoft, 2020-11-11:
Ok. Also we could always make 0.18.1

foosel, 2020-11-11:
Yeah... all that would change there would be the preinstalled OctoPrint version then, no need for a lengthy RC phase for that. Right now the testing of 0.18.0 is really more to iron out any kind of issues with the new base image and any changes done to OctoPi outside of OctoPrint, as far as I see it.

cp2004, 2020-11-11:
> Sending an M997 command...
> @sohxet 

Could this be related to https://github.com/OctoPrint/OctoPrint/issues/3683 since OctoPi 0.18 is likely now pulling in this upstream kernel update?

XRyu, 2020-11-11:
is it possible the arm64 does not come with raspi cam support?

````
Starting up webcamDaemon...

--- Configuration: ----------------------------
cfg_file:      /boot/octopi.txt
camera:        raspi
usb options:   -r 640x480 -f 10
raspi options: -fps 10
http options:  -w ./www-octopi -n --listen 127.0.0.1

Explicitly USB device:
-----------------------------------------------

Found video devices:
/dev/video0
/dev/video10
/dev/video11
/dev/video12
/dev/video13
/dev/video14
/dev/video15
/dev/video16
raspi
config file='/boot/octopi.txt':Start MJPG-streamer with video device: raspi
<13>Nov 11 17:42:15 root: Starting Raspberry Pi camera
Checking for VL805 (Raspberry Pi 4)...
  - It seems that you don't have VL805 (Raspberry Pi 4).
    There should be no problems with USB (a.k.a. select() timeout)
Running ./mjpg_streamer -o output_http.so -w ./www-octopi -n --listen 127.0.0.1 -i input_raspicam.so -fps 10
MJPG Streamer Version: git rev: 85f89a8c321e799fabb1693c5d133f3fb48ee748
ERROR: could not find input plugin
       Perhaps you want to adjust the search path with:
       # export LD_LIBRARY_PATH=/path/to/plugin/folder
       dlopen: input_raspicam.so: cannot open shared object file: No such file or directory
Done bring up all configured video device

Goodbye...
````

sohxet, 2020-11-11:
> > Sending an M997 command...
> > @sohxet
> 
> Could this be related to [OctoPrint/OctoPrint#3683](https://github.com/OctoPrint/OctoPrint/issues/3683) since OctoPi 0.18 is likely now pulling in this upstream kernel update?

That sounds exactly like my problem. I'll try to get any useful information from the pi and open a ticket upstream.

rbrian, 2020-11-12:
Are there any known backup./restore issues? i have a backup from my octoprint before the upgrade, and trying to restore it I get the "Uploading please wait" dialog for at _least_ 12 hours. Not going to let it go any further, will be easier to just redo things from scratch.

441Benny, 2020-11-12:
i have loaded/booted octopi 18 on my raspi 4 8gb worked great until i rebooted with desktop view now only seeing black screen with white dash top left corner wont let me type any commands though

Will-wastelander, 2020-11-12:
I have been having random drops where the system becomes inaccessible all together via SSH and WebGUI, but OctoDash is still able to access it ? When connecting via hardwire, no DHCP is pulled at all. I do not have a HDMI connected to it, but I do have a 7" screen, ask I am running OctoDash as previously mentioned. 

What info can I provide to assist with trouble shooting ?

rPi 4 4GB,, OctoPi 0.18.0 RC1 armhf, OctoDash 2.1.1, 7" DSI screen. This is the 3rd time it has happened in 24 hours.  However my  OctoPi 0.17.0 armhf, and 0.18.0 arm64 seem to be working fine. 0.17.0 I have no issues w/ network drops at all on the same system(s).

Will-wastelander, 2020-11-12:
Plugging in a Wireless USB kb/mouse appears to do nothing also. ctrl-alt-del does nothing, ctrl-c does nothing.

CRCinAU, 2020-11-12:
> 64bit amd64/aarch64

I'm sure you mean: 64bit arm64/aarch64?

guysoft, 2020-11-12:
> I'm sure you mean: 64bit arm64/aarch64?

Yea my bad, fixed. Its really a pain that its only one letter difference.

CRCinAU, 2020-11-12:
> Its really a pain that its only one letter difference.

Hahahah - tell me about it - that's why I call things aarch64 and x86_64 - even though debian loves to use amd64 :\

CRCinAU, 2020-11-12:
Hmmm - here we go - trying to use the HLS webcam stuff:

```
Nov 12 09:02:07 ender3.local ffmpeg[2381]: [h264_omx @ 0x558d4d4870] libOMX_Core.so not found
Nov 12 09:02:07 ender3.local ffmpeg[2381]: [h264_omx @ 0x558d4d4870] libOmxCore.so not found
Nov 12 09:02:07 ender3.local ffmpeg[2381]: Error initializing output stream 1:0 -- Error while opening encoder for output stream #1:0 - maybe incorrect parameters such as bit_rate, rate, width or height
Nov 12 09:02:07 ender3.local ffmpeg[2381]: [video4linux2,v4l2 @ 0x558d4c9de0] Some buffers are still owned by the caller on close.
Nov 12 09:02:07 ender3.local ffmpeg[2381]: Conversion failed!
Nov 12 09:02:07 ender3.local systemd[1]: ffmpeg_hls.service: Main process exited, code=exited, status=1/FAILURE
Nov 12 09:02:07 ender3.local systemd[1]: ffmpeg_hls.service: Failed with result 'exit-code'.

```

guysoft, 2020-11-12:
@chudsaviet any insight on this? ^

rbrian, 2020-11-12:
Getting a black rectangle where the Pi cam is supposed to display. Worked fine in previous versions. Not sure where a log would be that would have any info.

chudsaviet, 2020-11-12:
@CRCinAU , looks like it can’t find RaspberryPi hardware encoder libraries. Are you using the 64 bit version? I have never tested on it, my patches were submitted before 64 bit version was added.

XRyu, 2020-11-12:
on armhf version raspi camera works fine with mjpeg streamer. on arm64 i failed with the errors mentioned earlier :)

CRCinAU, 2020-11-13:
> @CRCinAU , looks like it can’t find RaspberryPi hardware encoder libraries. In RaspberryPiOS they are supposed to be installed as dependencies of ffmpeg.
> Are you using the 64 bit version? I have never tested on it, my patches were submitted before 64 bit version was added.

Yeah - this is on the 64 bit version.

Will-wastelander, 2020-11-14:
Is anyone else getting random wifi drops w/ armhf ? It happens to me about every other day right now..

foosel, 2020-11-14:
I observed something that felt like a WiFi drop easier this week during the final update tests for 1.5.0rc1, but I was under a deadline and this couldn't investigate what it was.

Just putting it out there that yes, saw something that could qualify.

It reminded me of the times when wifi power management was broken and the devices would just shut down WiFi.

Will-wastelander, 2020-11-14:
Gina,

This is more like a system hang. As eth0 doesn't pull dhcp when a network
cable is connected during this issue. The rPi is in an enclosure, but temps
are 117f max from what I have seen so far. That temp shouldn't cause any
issues, that I know of. I dont have any output of an error on my DSI
connected screen, running OctoDash 2.1.1 either.

foosel, 2020-11-14:
As I said, I didn't have time to investigate. No chance to test wired either. I can confirm one incidence of "pi connected via WiFi no longer responded to pings", a reboot fixed it.

Will-wastelander, 2020-11-14:
Gina,

Correct, reboot resolves issue. Is there anything I can look at for logs,
etc to provide to assist with trouble shooting ?

nes-jisc, 2020-11-15:
> I have been having random drops where the system becomes inaccessible all together via SSH and WebGUI, but OctoDash is still able to access it ? When connecting via hardwire, no DHCP is pulled at [...]

I have seen this same issue with my wifi connected Pi. Octodash continues to work, printing continues but no connection to octoprint web interface or SSH. I am switching to a wired connection to see if this makes any difference.

chudsaviet, 2020-11-16:
Regarding HLS-on-arm64:
1. There are no VideoCore libraries in `/opt/vc/lib`. Fixes by `rpi-update`, not by Apt, because these are proprietary-ish libraries.
2. FFMpeg is built with different flags by default. It needs to have `--enable-omx-rpi` to work with `h264_omx` encoder on Rpi.

I tried to build from source, but it couldn't find the libraries despite they were existing. Will continue work later.

mordhau5, 2020-11-16:
Does anyone know if Python2 is removed at the system level?

chudsaviet, 2020-11-17:
@mordhau5 , even if it is, you can always install it fro APT.

mordhau5, 2020-11-17:
@chudsaviet true, I guess my real question is if it's going to keep being supported by raspbian, but there's no way they'd pull that so soon.

guysoft, 2020-11-17:
Python 2 has reached end-of-life on the start of 2020. So it will not eventually
https://www.python.org/doc/sunset-python-2/

mordhau5, 2020-11-17:
@guysoft yep. Hopefully Klipper team makes moves on this before raspbian drops it completely.

chudsaviet, 2020-11-23:
Created bug https://github.com/guysoft/OctoPi/issues/697 to track HLS-on-arm64 issues.

dualznz, 2020-11-28:
Stable on Pi 4 8gb 32-Bit

guysoft, 2020-11-29:
@chudsaviet Is this a show stopper for 0.18.0? Do you think you can fix it and we shall release an RC2?

chudsaviet, 2020-11-29:
@guysoft , I don’t think its a show stopper. We are advertising HLS as experimental, so the fact its not working on arm64 is not blocking the release I think.

jneilliii, 2020-11-29:
@guysoft with the release of octoprint 1.5.0 right around the corner, maybe we should have an rc2 round with that version once it comes out stable and then go stable with 0.18 shortly after?

Will-wastelander, 2020-11-30:
Is there any update on what is causing the wifi to completely stop working on the ARM64 build ? I've offered to provide info, but no one has provided me details on how to get the info that would be needed.

guysoft, 2020-11-30:
@Will-wastelander Currently I am not sure how to debug this at all. It might be worth reporting to https://github.com/raspberrypi/linux . But I doubt they will take anything under condiration with out some for of log. Perhaps the output of ``dmesg`` after the network is down should be enough to understand what is going on.

You can save dmesg to a file using ``dmesg > /home/pi/dmesg.log``  with a screen and keyboard attached, and then reboot, copy it off and send.

Will-wastelander, 2020-11-30:
Guysoft thats will be impossible to do, as plugging in a keyboard and mouse
doesn't work. Once the system is hung, there is nothing I can do, other
than reboot. I will try to do this again, but the last 3 times I tried,
nothing worked.

Will-wastelander, 2020-12-01:
@guysoft I also don't seem to have the issue w/ my rasbianos arm64 systems.

ideal2545, 2020-12-01:
This has been working really well for me, any word on when its going offiical release?

gdombiak, 2020-12-01:
Successfully tested HLS using OctoPi 0.18.0 RC1 and OctoPrint 1.5.0 (released version). I was able to see the camera in OctoPrint webUI and also in OctoPod (unreleased version). However, I noticed something that wanted to bring up here in case it was not discussed before.

I'm using a Raspberry Pi 3 Model B Plus Rev 1.3 with a single OctoPrint instance (no plugins) running and using a RPi camera v2. CPU usage is extremely high and OctoPrint is doing nothing. It is not even connected to a printer. I have no client connected to OctoPrint and nothing is connected to the camera.

[image]

Is there a warning somewhere indicating that using HLS with a RPi 3 might have an impact on print quality when printing from OctoPrint? I have not tested this so cannot confirm that it will have an impact. However such high CPU usage (with minimal load) makes me wonder how much CPU is going to be left for OctoPrint and potential impact when printing from OctoPrint.

I purchased a couple RPi 4 where I will have this setup for my local "farm" of printers. Having a more powerful CPU, I anticipate that it should not be an issue with RPi 4. 

Anyway, too long of a post to ask: Has anyone tested HLS with RPi 3B (+?) and printed from OctoPrint? Have you observed any impact on print quality? Is there value adding a warning that RPi 4 is recommended when enabling HLS?

Thanks,
Gaston

jneilliii, 2020-12-01:
I think that's why it's labeled as experimental. I too am running this combination and have only printed once since enabling hls and did not see any print quality issues, but it was a simple print.

chudsaviet, 2020-12-01:
Hi, @gdombiak .
Its nice to see people start testing HLS feature :) 
Unfortunately, high CPU usage is not going to go anywhere. Even with the fact we are using hardware h264 encoder, significant CPU is required, both for HLS and high FPS JPEG encoding (there is a side JPEG stream for timelapses).
Good news are that 122% in ‘top’ on a 4-core machine is not a very high usage. Your capacity is 400%.
Moreover, we run FFmpeg with lower process priority, so its supposed not to displace OctoPi. On the other side, Linux is not a real time OS, so its not guaranteed.

Please, try to do test prints, the community will be really glad if you do some real life testing. :)

gdombiak, 2020-12-01:
@chudsaviet, thanks for sharing that info. That feels better knowing that it runs with a lower priority. When I monitored a while while while back CPU usage when using MJPEG (when I installed multiple webcams on the same RPi 2) I noticed that if no one was watching the camera then there was no CPU impact/usage. I was expecting to see something similar with HLS. Anyway, once I released the new version of OctoPod (iOS, AppleWatch and AppleTV) with HLS support I will go back to printing and test how things work.

@jneilliii hey man! Nice to see you around this repo as well! :) When I read that it was experimental I assumed that it might break/fail. If we confirm that it introduced print quality issues (based on @chudsaviet comment I would expect to NOT be the case) then a more explicit warning might be needed or maybe a recommended hardware requirements message might help. Anyway, I'm getting ahead of ourselves. First let's confirm this is an actual issue.

[47 more replies]

closed 2021-01-11 by guysoft