guysoft/OctoPi #694: Cannot SSH into a clean install of OctoPi 0.17.0

rlgjr562, 2020-11-08:
This issue has been reported before and marked closed.
 I just installed a clean copy of 0.17of OctoPi and I cannot connect with SSH. 

I see in past comments on this issue to use the command sudo raspi-config which I have done many times with a normal Raspberry PIs (one with a keyboard and Monitor).  The issue is many OctoPi are headless. The root issue needs to be found on why SSH is not configured  or a way to run the command or enable SSH from within the OctoPrint software.  Maybe OctoPi needs a plugin that allows access to the underlying OS.

rlgjr562, 2020-11-09:
I found a way to get  ssh installed, by simply putting a file named SSH in the boot director and bounce the Pi,  Could OctoPi do this as part of the standard 0.17.0 build?

guysoft, 2020-11-10:
It is part of the standard build, this is the line that does it:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/base/start_chroot_script#L63

I am not sure what you have done with your system, but that is not standard OctoPi

rlgjr562, 2020-11-11:
It was fresh down load load and install of the octopi image 0.17.0.  It was a standard build but ssh did not seem to installed  or running.  So I punted and I found the ssh file thing in the pi community boards so I could get it working.  Ignore my ssh file suggestion but please fix why SSH is not working by default on a fresh install.

guysoft, 2020-11-16:
Does it happen on 0.18.0RC1?

inthroxify, 2020-12-02:
> Does it happen on 0.18.0RC1?

This is happening to me with 0.17.0 and with 0.18.0RC1.

Wrote 0.17 to the SD card with Rufus on Windows 10. The pi boots, and can connect via ssh with putty, but attempting to log in with pi/raspberry just returns an access denied error.

Did all of the same with 0.18.0RC1, with the exact same result. The ssh file is present in the root of the SD card when "mounted" on Windows.

coaxial, 2021-07-17:
This is happening to me as well with a fresh Octopi install and a Pi 4 on the latest firmware. Using octopi.local or the IP shown in the boot console makes no difference.

guysoft, 2021-07-17:
Can you write paste/screenshot what kind of error you are getting?

inthroxify, 2021-07-17:
Also, what make/model is your WiFi access point, if you don't mind me asking? And does the same problem occur while using hard-wired Ethernet?

coaxial, 2021-07-24:
I was not getting any error on the Pi, just not able to connect to it from another machine on the same network. My AP is a UniFi AP-AC-Lite.

I don't know what happened, it just wouldn't accept any connection either via SSH or HTTP. I then erased the SD card and wrote the image to it again, and it worked. I was not able to reproduce the issue after that.

inthroxify, 2021-08-02:
>  My AP is a UniFi AP-AC-Lite.

Interestingly, that is the same access point that I have, and we both had this issue. @coaxial keep an eye out to see if your wifi drops after a few days/weeks and then refuses to reconnect until you reboot the Pi; that is the issue I am facing now.

guysoft, 2021-08-03:
Interesting that its the same access point.
I can see also people complaining about this here:
https://community.ui.com/questions/WIFI-Clients-losing-network-connectivity/9de0d0ea-7722-45c6-9709-a7815ffc0298