guysoft/OctoPi #698: Less CPU load for video capture.

posicat, 2020-11-24:
Working on a project to automate my garage door I found that I was able to get better streaming with less CPU usage using ustreamer instead of mjpg_streamer.  Just thought I'd pass that along, as it leaves more CPU free for other functions.

https://github.com/pikvm/ustreamer

reedy, 2020-11-24:
Is ustreamer packaged for debian (or derivatives) anywhere? and/or via a PPA/similar? Quick google suggests not, which might be a bit of a blocker for doing this...

And for reference about the benefits of it (used in other projects than just pi-kvm) - https://mtlynch.io/tinypilot/#ustreamer-a-super-fast-video-streamer

tedder, 2020-11-24:
doesn't look like debian, but arch and Docker; the latter shows the build chain, and it might be handy to start building up interoperable containers for pi4. (just my thought)

tinypilot talks about the latency win for it.

reedy, 2020-11-25:
I imagine it might not be too hard to find a Debian Developer to help get it in for a future release (and then backporting probably becomes a bit easier)...

I just noticed https://github.com/pikvm/ustreamer#building

reedy, 2021-02-03:
> ustreamer_3.12-1_amd64.changes ACCEPTED into unstable, unstable

It should mean it's part of Debian Bullseye (11)..

reedy, 2021-03-25:
And it's in Raspbian Testing now....

reedy, 2021-03-25:
https://tracker.debian.org/pkg/ustreamer
https://packages.debian.org/bullseye/ustreamer
https://repology.org/project/ustreamer/versions

vincele, 2022-01-07:
It's in armbian stable also now:

```
yo@opi:~# cat /etc/debian_version 
11.2
yo@opi:~# apt info ustreamer
Package: ustreamer
Version: 3.16-1
Priority: optional
Section: video
Maintainer: Sam Reed <sam@reedyboy.net>
Installed-Size: 307 kB
Depends: libbsd0 (>= 0.6.0), libc6 (>= 2.17), libevent-2.1-7 (>= 2.1.8-stable), libevent-pthreads-2.1-7 (>= 2.1.8-stable), libjpeg62-turbo (>= 1.3.1), libuuid1 (>= 2.16)
Homepage: https://pikvm.org/
Download-Size: 104 kB
APT-Sources: http://deb.debian.org/debian bullseye/main armhf Packages
Description: Lightweight and fast MJPG-HTTP streamer
```

guysoft, 2022-01-08:
Nice, so I guess we could look in to adding it in the next release. If anyone wants to try and make a PR that could speed things up