guysoft/OctoPi #699: 0.17.0 haproxy issue

AaronAnz, 2020-12-02:
#### What were you doing?

1. Install 2018-11-13-octopi-stretch-lite-0.16.0 or greater
2. Configure network
3. visit ip address in web browser

#### What did you expect to happen?
to see web panel

#### What happened instead?
refused to connect

#### Did the same happen when running OctoPrint in safe mode?
not sure

#### Version of OctoPi
0.17.0 or greater

I have a raspberry pi 3B+ and it is unable to work out of the box. I know how to configure networking on the device and verified it was running. HAProxy is not running on its own and will not start at all. I had to install 0.16.0 and it worked flawlessly.

closed 2022-08-20 by AaronAnz

guysoft, 2022-08-21:
Thanks for reporting the issue was resolved :)

AaronAnz, 2022-08-21:
It was never resolved. I don't even use octopi anymore. Was just going through old issues I had open on various libraries and closed it since it was 2 years old and no one even responded.

guysoft, 2022-08-21:
Well its a little late to debug it now, the debug otuput from haproxy would have helped to understand what is going on, it might not have even been HAProxy but OctoPrint reporting something.

AaronAnz, 2022-08-22:
It's not like anyone ever even responded requesting more information. It was a fresh install at the time and just straight up broken. You clearly didn't care about it.

guysoft, 2022-08-22:
Sorry dude, I really go this as volunteer work and sometimes I really don't have time to fix stuff. I do what I can.