guysoft/OctoPi #700: Webcam not appearing for 64bit 0.18 nightlies

Taomyn, 2020-12-05:
#### What were you doing?

Testing latest 64bit 0.1.8 nightly (2020-12-05_2020-08-20-octopi-buster-arm64-lite-0.18.0) on my Pi 4

#### What did you expect to happen?

Pi camera should appear

#### What happened instead?

Pi camera is not appearing

#### Did the same happen when running OctoPrint in safe mode?

Not relevant

#### Version of OctoPi

2020-12-05_2020-08-20-octopi-buster-arm64-lite-0.18.0
2020-12-04_2020-08-20-octopi-buster-armhf-lite-0.18.0

#### Printer model & used firmware incl. version

Not applicable

#### Screenshot(s)/video(s) showing the problem:

[image]

#### Other comments

I am currently running an older build of 32bit 0.18 from some months ago and wanted to test the 64bit version. The past few times I have tried this the Pi camera does not appear to be working even though it was fine with my running build. Everything else appears to be fine. I have now tried with the latest 32bit nightly and the Pi camera is detected like before, so this is specific to the 64bit version.

I also switched cameras from a original Pi camera v2.1 to a new clone model that provides night-vision and it's doing the same.

From 64bit:
[webcamd.log](https://github.com/guysoft/OctoPi/files/5647116/webcamd.log)
[image]

From 32bit:
[webcamd_32.log](https://github.com/guysoft/OctoPi/files/5647121/webcamd_32.log)

From both, as curious about message:
[image]

[image]

guysoft, 2020-12-06:
Did you try a nightly or the RC1? #692

Taomyn, 2020-12-06:
Nope, just the ones directly from the nightly download area.

The 32bit nightly version continues to run well btw.

guysoft, 2020-12-06:
Ok, would appreciate it if you check the RC1, since that is what is being planend to release. It was also teststed by other people that did not report such an issue.

Taomyn, 2020-12-06:
Still the same, totally fresh install after amending the config files for my camera and wifi:
[image]
[image]

guysoft, 2020-12-06:
Ok, the last screen shot helped me locate the issue. 
Its being handled here: https://github.com/jacksonliam/mjpg-streamer/issues/259
(This is the streaming we are using).

The only workaround I can think of is using the HLS streaming solution that @chudsaviet is working on, that also has some issues on arm64 (see https://github.com/guysoft/OctoPi/issues/697 ).

@chudsaviet have you had any success that might work here?

chudsaviet, 2020-12-06:
If the camera is not seen by the system - HLS will not work either, because it uses the same V4L2 input.

@Taomyn , check next things:
1. Check physical connections. I know this advice is trivial, but you wont believe how many people miss this thing :) 
2. Try run raspi-config and enable webcam.

Taomyn, 2020-12-06:
@chudsaviet if I use anything except the 64bit build it does not work - it's all above in the OP. Same Pi, same camera + cable in fact same two cameras as I also have an official v2 Pi-Cam. A clean 32bit version without running raspi-config which is how I have been testing shows the same difference and the cameras work, it's just the 64bit version that is failing.

chudsaviet, 2020-12-06:
@Taomyn , default config can be different in arm64 build. Try raspi-config.

Taomyn, 2020-12-06:
@chudsaviet no difference. The error message reported is pointing at missing files not that the camera isn't present:

```
MJPG Streamer Version: git rev: 85f89a8c321e799fabb1693c5d133f3fb48ee748
ERROR: could not find input plugin
       Perhaps you want to adjust the search path with:
       # export LD_LIBRARY_PATH=/path/to/plugin/folder
       dlopen: input_raspicam.so: cannot open shared object file: No such file or directory

```

chudsaviet, 2020-12-06:
Oh, I see. Maybe mjpg-streamer don't have the library built.

antoneliasson, 2020-12-18:
Not sure if it is still unclear whether the camera is supposed to work on aarch64 so I tried 0.18 RC1 aarch64 briefly on a RPi 4 (4 GB) with an official Raspberry Pi v2 camera module connected. It worked flawlessly out of the box. While streaming MJPEG at 640x480 haproxy uses about 2 % of one CPU core and mjpg_streamer about 1 %.

/root/bin/webcamd selects input_uvc.so instead of input_raspicam.so. input_uvc uses V4L2 which works on aarch64 unlike input_raspicam which uses the Broadcom specific 32-bit only MMAL interface.

1n5aN1aC, 2020-12-18:
Same situation here.

Working system, installed 0.18 rc1 (64bit) because I needed Python3, and restored from backup.  raspi webcam does not work.

webcamd.log has the same:
```
MJPG Streamer Version: git rev: 85f89a8c321e799fabb1693c5d133f3fb48ee748
ERROR: could not find input plugin
       Perhaps you want to adjust the search path with:
       # export LD_LIBRARY_PATH=/path/to/plugin/folder
       dlopen: input_raspicam.so: cannot open shared object file: No such file or directory
```

I ran `vcgencmd get_camera` and got `supported=1 detected=1` so it should be connected and working....

1n5aN1aC, 2020-12-18:
Update:
- Flashed 64bit RC1 image, webcam works.
- Updated OctoPrint to 1.5.2, webcam works.
- Restored Octoprint from backup:  webcam works.

I need a working printer relatively soon, so I won't be testing `sudo apt update && sudo apt upgrade` for now, but I somewhat suspect that is what killed it in my case at least.

CobaltEcho, 2021-02-05:
Did this ever get resolved, or a solution found? I have the v2 PiCam was banging my head against the wall for 2.5hrs yesterday before I came across this, lol.

guysoft, 2021-02-07:
@CobaltEcho , no, this is stuck here: https://github.com/jacksonliam/mjpg-streamer/issues/259

Also, when you download a nightly build its at your own risk, its untested and is not guaranteed to work.
I know usually they do, however, the 64bit builds, specifically the raspberypi foundation ones have been abandoned in favour of Ubuntu, because they were too unstable.

chudsaviet, 2021-02-07:
“the raspberypi foundation ones have been abandoned in favour of Ubuntu” - wut? Any proofs?

guysoft, 2021-02-07:
@chudsaviet See: https://github.com/guysoft/OctoPi/issues/711#issuecomment-764612030

and: https://github.com/RPi-Distro/pi-gen/issues/481#issuecomment-745153854

You can also see the images are not updated here: https://downloads.raspberrypi.org/raspios_lite_arm64/images/
They made one build in August 2020, since then there were 3 released which did not include 64bit.

chudsaviet, 2021-02-07:
It doesn’t mean “abandoned”, it means “beta”. So in future they will support 64bit builds.

CobaltEcho, 2021-02-08:
@chudsaviet , I have a normal USB Webcam I can use for the time being.  Bit maybe this is a stupid questions, but looking at the list of nightlies...  How do I find RC1 or RC2 (I think that's out) for 64bit?  

http://unofficialpi.org/Distros/OctoPi/nightly-arm64/?C=M;O=D

guysoft, 2021-02-08:
There is no RC2 for 64bit.
I released an RC1 and it was clear its not working.  So it was not released and there was no RC2.

CobaltEcho, 2021-02-08:
I tried the nightly build from 11 Jan 2021 and the Pi Cam came right up,
and have had no issues with it.

guysoft, 2022-01-01:
Hey, updating that I also have a 64bit OctoPi running with a working rpi webcam.