guysoft/OctoPi #701: [Brainstorm] Adding HLS options to `webcamd` or creating `hlsd`?

chudsaviet, 2020-12-07:
Hi!
It looks like I will have to add more options and logic to HLS streaming, so I will need a script.
I see two options:
1. Add HLS code to `webcamd`, but it may make the script a spaghetti code, because HLS options are significantly different from MJPG-Streamer options.
2. Create a separate script for HLS - `hlsd`. I would prefer to use Python 3 for this. Disadvantages are to support two code bases which have partially similar logic.

I'm leaning towards option 2.

chudsaviet, 2020-12-07:
@guysoft , @foosel , please share your opinions on this topic.

guysoft, 2020-12-07:
I think I go for option 2 as well. You already have python3 and it should make the code more readable for future people contributing to this.
It would also make it easier for a user to see which service is running using ``ps aux`` or ``systemctl status``, you will see a log.

It might even make the code portable for other CustomPiOS distros, if written in a separate module.

guysoft, 2020-12-07:
Note:
If you need python3 site packages, it might be a good idea to use the system virtualenv and not OctoPrint's virtualenv

chudsaviet, 2020-12-08:
Thank you, @foosel , @guysoft .