guysoft/OctoPi #702: [0.18 RC1 - 64-bit build]: OSError: libmmal.so: cannot open shared object file: No such file or directory

kennethjiang, 2020-12-08:
#### What were you doing?

I'm the author of The Spaghetti Detective and OctoPrint Anywhere plugins. Users of both plugins have run into the same issues when they are using 0.18 RC1.

Here are the minimum number of steps to reproduce the issue:

1. Flash an SD card with 0.18 RC1, 64-bit build. Do not run any other commands such as `sudo apt upgrade`.
2. `source ~/oprint/bin/activate`
3. `pip install picamera`
4. `python -c "import picamera"`

#### What did you expect to happen?

The command runs successfully with no output.

#### What happened instead?

The command outputs:

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/pi/oprint/lib/python3.7/site-packages/picamera/__init__.py", line 72, in <module>
    from picamera.exc import (
  File "/home/pi/oprint/lib/python3.7/site-packages/picamera/exc.py", line 41, in <module>
    import picamera.mmal as mmal
  File "/home/pi/oprint/lib/python3.7/site-packages/picamera/mmal.py", line 49, in <module>
    _lib = ct.CDLL('libmmal.so')
  File "/usr/lib/python3.7/ctypes/__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libmmal.so: cannot open shared object file: No such file or directory
```

#### Did the same happen when running OctoPrint in safe mode?

N/A

#### Version of OctoPi

0.18 RC1. Please note this bug only happens in the 64-bit build. The 32-bit build is good.

#### Printer model & used firmware incl. version

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

For a bit more info: `libmmal.so` is the driver for Pi Camera. On older versions of OctoPi, as well as the 32-bit build of 0.18 RC1, it is located in `/opt/vc/lib`. However, the entire folder is missing from the 64-bit build of 0.18 RC1.

kennethjiang, 2020-12-08:
This issue may be the duplicate of https://github.com/guysoft/OctoPi/issues/697

guysoft, 2020-12-08:
Also related to #700

guysoft, 2021-01-11:
Now also related to #711

guysoft, 2021-06-19:
The now Ubuntu 64bit builds might solve this. I am not sure

kaczmaro, 2021-12-31:
Did you solve it? having the same issue :(

guysoft, 2022-01-03:
@kaczmaro Please provide the image you are using. There is a nightly build that might solve it.

gcd0318, 2023-02-04:
I'm using Raspberry Pi OS Lite (64-bit) Release date: September 22nd 2022, and python3-picamera/stable,now 1.13 arm64, and got the same error

While, I found a PiCamera2, and it seems able to import. I'm still trying how to use it. It PiCamera2 relative with PiCamera?