guysoft/OctoPi #707: Securing an OctoPi: What measures can I take?

shueja-personal, 2021-01-01:
Hello,
This request is probably going to be a bit different than other security-related questions. Instead of trying to safely expose my printer to the Internet, I'm trying to make sure that it can't be accessed at all except through a few specific methods from another device on the local network. I'm doing this on my built-from-source OctoPi (and auto-hotspot, and a custom OctoPrint plugin) distro. I'm seeing this link:
https://www.raspberrypi.org/documentation/configuration/security.md

What measures out of these are incompatible with OctoPi? I'm specifically considering shutting down SSH and terminal access except by a "danger zone" switch in my plugin, blocking all incoming network access besides from the local network (my distro would still need to access certain external sites for things such as automatic plugin updates), and potentially removing the pi user and replacing it with something else. This last measure I am unsure about, because so much of the OctoPi and CustomPiOS code seems to depend on a "pi" user. Is it sufficient to run ```sudo deluser pi``` and keep the /home/pi directory?

What other security measures would you recommend for making sure the Pi and the printer can only be accessed from the local network?

DrDrache, 2021-01-03:
best to ask here :
https://community.octoprint.org/

guysoft, 2021-01-03:
1. You should have in your router settings of a firewall. I use OpenWRT for this kind of stuff. You should have a WAN and LAN zone. In general this is good practice and it required if you want a security environment.
2. Take a look at ufw firewall (its called Uncomplicated Firewall, hope its not that hard, I use that too).

shueja-personal, 2021-02-06:
Thanks for the response.

closed 2021-02-06 by shueja-personal