guysoft/OctoPi #710: OctoPi 0.18.0 RC2 Status

guysoft, 2021-01-11:
First release candidate for OctoPi 0.18.0

~~There are both 32bit and 64bit Based on RpiOS images available.~~ 64bit is unstable and is delayed for 0.19.0. We have nightly builds for it if you really need it.

* Support to Raspberry Pi 4B 8GB and Raspberry Pi 400.

* The main change that would be likely noticeable to plugin developers is that OctoPi comes with Python3 now.

Changes from RC1:

* OctoPrint 1.5.2

Please try the release candidate.

32bit armf:
Download it at:
http://unofficialpi.org/Distros/OctoPi/2021-01-11_2020-12-02-octopi-buster-armhf-lite-0.18.0.zip

Md5: [``43387c99873210969a21083520ec963b``](http://unofficialpi.org/Distros/OctoPi/2021-01-11_2020-12-02-octopi-buster-armhf-lite-0.18.0.zip.md5).

64bit arm64/aarch64:
Not this release, use nightlies

### Changes in the image

* OctoPrint 1.5.2
* OctoPi now uses Python3
* ``/home/pi/scripts/safemode`` (thanks @OutsourcedGuru )
*  Fix #619 Shutdown requires network (Thanks @da4id )
* #623 from szopen111/fix-issue-621 (Workaround for low resolution cameras)
* Fix/enable haproxy logging #640 and enable compression at haproxy layer #643 (thanks @tedder )
* Experimental HLS support #650 #668 #688 (thanks @chudsaviet )
* Removed old Cura.  See #654 and #444 .
* Some usb cameras setting fixed (#657 https://github.com/guysoft/OctoPi/commit/b64a5441b807f449417b8bcedb41b2703adf89e9 https://github.com/guysoft/OctoPi/commit/0d0ea3751e3866d7f893699894691025e33a7757 )
*  Fixes #676 and adds deps to install numpy from wheel for plugins taht need numpy
* New arm64 builds
* Set HAProxy Diffie-Hellman key size to 2048 #685

### Build notes
* Uses [CustomPiOS 1.4.0](https://github.com/guysoft/CustomPiOS/releases/tag/1.4.0)

foosel, 2021-01-11:
[Blog post published](https://octoprint.org/blog/2021/01/11/octopi-release-candidate-0-18-0rc2/)

MaximilianGaedig, 2021-01-11:
how, if I can, can I update easily from 17.0?

cp2004, 2021-01-11:
As with all OctoPi releases, you have to re-flash your SD card. Easiest way would be to backup your OctoPrint instance and then restore this backup on the new image.

Tecnoid69, 2021-01-11:
> As with all OctoPi releases, you have to re-flash your SD card. Easiest way would be to backup your OctoPrint instance and then restore this backup on the new image.

Do you suggest to do the same to upgrade from OctoPi 0.18.0 RC1?

jneilliii, 2021-01-11:
yes, that would be the same method.

Tecnoid69, 2021-01-11:
> yes, that would be the same method.

But with the restore I'll found all the settings? (OctoPrint, OctoScreen...and so on?)

cp2004, 2021-01-11:
OctoPrint yes, everything else no.

MaximilianGaedig, 2021-01-11:
Will the backup restore plugins, also will the plugins still work? Will python 2 remain? Or will all plugins break bc they still use python 2?

cp2004, 2021-01-11:
Backup restores plugins, it reinstalls them to the system. You can check using the plugin [Python 3 Compatibility check](https://plugins.octoprint.org/plugins/Python3PluginCompatibilityCheck/).

For further questions or support, please use the [community forums](https://community.octoprint.org) or [discord server](https://discord.octoprint.org), to keep this issue reserved for discussion about the release candidate itself rather than support questions. This is the last one we're answering here. Thanks.

Malvu, 2021-01-13:
I am testing the OctoPi Release Candidate 0.18.0rc2 version, the printer systematically fails at an irregular interval.
Creatily Ender 3 Pro
Motherboard 4.2.2
Marlin vers 1.1.2

guysoft, 2021-01-13:
@Malvu What error is it throwing back?

jneilliii, 2021-01-13:
This seems more of an issue with these parameters...
```
Creatily Ender 3 Pro
Motherboard 4.2.2
Marlin vers 1.1.2
```
I recommend you read up on this...with possible fixes.

https://community.octoprint.org/t/printers-known-to-ship-with-broken-or-unsafe-firmware-or-broken-electronics/25911?u=jneilliii

cp2004, 2021-01-13:
It's not an OctoPi issue, their post here https://community.octoprint.org/t/critical-resend-ratio/28914/2 - it happened on other versions too.

AlexSwensen, 2021-01-15:
I had an issue trying to download from the link. I ended up having to download with `wget`.

Just a heads up. If anyone has any issues, I have uploaded a mirror [here](https://drive.google.com/file/d/1UF2eQmpfYIYCSoJHpZ4Cnk-L6gatTTKJ/view?usp=sharing)

evenprimes, 2021-01-17:
I've been running the image the last couple of days. The issue I'm having is that I can't seem to get MQTT working. I've got it feeding into MQTT for a little monitoring and automation with Home Assistant. 

Resetting the MQTT config in Octoprint doesn't bring it back up, and flushing MQTT setup in HASS only bring back my non-Octoprint entities.

I installed on a fresh CD card and restored from an Octoprint backup also on 1.5.2, on an Octopi 0.17 install.

Octoprint info:
```
browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:84.0) Gecko/20100101 Firefox/84.0
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1200
env.hardware.ram: 915722240
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 3 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.2
```

mike-bee, 2021-01-18:
I am running rc2 for 5 days now with a couple of prints. Restore from 17.0 with OctoPrint 1.5.2 worked right away. Used and restored plugins include Filament Manager, Firmware Updater, Navbar Temperature, Octolapse, Printoid Notifications and WiFiStatus.

No negative observations so far.

System Info:
browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.141 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1200
env.hardware.ram: 915722240
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 3 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.2

bvrielink, 2021-01-19:
FYI: upgraded from 0.17 to 0.18 rc2 and everything works correctly (restoring backup, printing, usb camera etc). Also didn't find any problems with the 15 plugins I am using. Running on a Raspberry Pi 3 B+ btw.

DroneMang, 2021-01-20:
Does this update work well with Klipper?  I want to try the new ArcWelder dev version, and it needs python 3 so might as well try this with it.  Thanks.

leigh-johnson, 2021-01-20:
:tada: Thank you for the work you do on OctoPi! 

I'm setting up smoke tests for https://github.com/bitsy-ai/octoprint-nanny-plugin against this distribution. :rocket: 

[image]

AlexSwensen, 2021-01-21:
Just an update. I have been running this the last few days with my Creality Ender 3 V2, and haven't had any issues.

MaximilianGaedig, 2021-01-22:
Anycubic i3 Mega S on an Rpi zero, already have this like 5-7 days, no problems so far.

M1dn1ghtN1nj4, 2021-01-25:
Custom RatRig V-Cast, Marlin Bugfix (as of 1-22-2021), SKR E3 Mini 2.0, RPi 4 4GB, OctoPi 0.18.0 RC2.  Runs perfectly.  I've been trying out new plugins and features.  So far, no hiccups.

guysoft, 2021-01-25:
Released: https://github.com/guysoft/OctoPi/releases/tag/0.18.0

closed 2021-01-25 by guysoft

cp2004, 2021-01-25:
@guysoft 
Might just be worth adjusting this comment in the release notes:

> 64bit is needed for fully support al the memory space for Raspberry Pi 4B 8GB

Since 64 bit is not really *needed* for anything usually done on OctoPi, since nothing really uses more than 3GB (iirc? hope that's right) memory per-process. We've had enough people confused that they think 64 bit is the *only* one that supports their Pi4 8GB, and since these are untested nightlies that at the moment don't quite work, I would avoid encouraging use of them for now.

guysoft, 2021-01-26:
@cp2004 Done.
Also - in theory the 64bit might also be helpful to add slicing on the Pi in future versions.